CANopenNode/CANopenNode #5: How to use

Dawud-Beale, 2016-01-18:
Hi, I am a bit of a newb on using CAN communication. I need a cost effective method of pulling CANopen data from a draw string sensor.

How can I use your source code to do this? What hardware will I need and what are the basic steps?

CANopenNode, 2016-01-18:
It's so general question, I can't answer much. It depends, what for you will need the data.
If you have a Linux computer (PC or Raspberry PI or Beaglebone or something), you can use a CAN interface supported by Linux kernel + CANopenSocket. Here you also have some CANopen master functionality.
If you prefer microcontrollers, you have an example for dsPIC33, etc in CANopenPIC.

There is a lot of CAN and CANopen related literature.

closed 2016-01-18 by CANopenNode

Dawud-Beale, 2016-01-18:
My application is a draw string sensor with an encoder attached to it. The encoder is a Baumer GXMMW and communicates sensor data via CANopen.

I am hoping to get the data into LabVIEW for displaying and closed loop control later on.

I am open to different methods of porting the data into LabVIEW as long as its cost effective.

An Arduino or PIC would be fine, unless its possible with a Compact DAQ. If there is any other alternative options you could recommend for hardware I am open minded as long as its cheap.

Do you have any recommendations?

Dawud-Beale, 2016-01-18:
My application is a draw string sensor with an encoder attached to it. The encoder is a Baumer GXMMW and communicates sensor data via CANopen.
I am hoping to get the data into LabVIEW for displaying and closed loop control later on.
I am open to different methods of porting the data into LabVIEW as long as its cost effective.
An Arduino or PIC would be fine, unless its possible with a Compact DAQ. If there is any other alternative options you could recommend for hardware I am open minded as long as its cheap.
Do you have any recommendations? 
Thanks and best wishes
Dawud Beale

CANopenNode, 2016-01-19:
You probably need CAN interface to LabVIEW. I don't have any other recommendations for it.