CANopenNode/CANopenNode #6: Clarification on GPLv2 with Classpath Exception vs. LGPL

JoelTheMole, 2016-02-05:
I am interested in using this library with a STM32F4XX microcontroller.  The project I am planning on using it with is a commercial project, and I would not be able to apply the GPL license to the commercial portion of the code.

I am trying to ascertain the correct approach to cleanly separating the commercial client code from the CANopenNode library code.  Is the exemption applied in a similar manner to that of the LGPL?  In other words, if I design my client code so that it only uses this library via linking, will that be allowed?  If so, are both static and dynamic linking allowed?

Of course the terms of the GPL will be maintained otherwise, and this library's source, along with any modification made in order to get it working on the microcontroller (driver development, etc.) will be made available to anyone using the product being developed.  It is also my intention to provide any updates to this code base back upstream via a pull request, presuming you are interested in such modifications.

Is this the correct approach?

CANopenNode, 2016-02-06:
I think so. See some discussion here:
https://sourceforge.net/p/canopennode/discussion/387151/thread/b7910450/

I don't know how reliable is code for STM32 now. So any revisions and pull requests are very welcome. It would also be nice to put a link to some very basic example with demo board and this microcontroller, similar as is CANopenPIC here.

JoelTheMole, 2016-02-08:
Great.  Thank you for your quick response.  In the end it is up to the customer if we use this code base, or another, but I do hope that I will have the opportunity to do so, and to contribute back what is developed.

closed 2016-02-08 by JoelTheMole