CANopenNode/CANopenNode #10: PIC32MX 1/2/5 devices only have 16 CAN filters

dmarcosgon, 2016-07-05:
Available CAN filters for PIC32 is hardcoded to 32 (see line 137 of CO_driver.c), while PIC32MX 1/2/5 only have 16. Trying to configure filters past the 16th fails and the associated COB-ID message is ignored. For example PIC32MX530F128L is affected. The datasheet points to the CAN section of PIC32 family reference manual which says there are 32 filters, but the datasheet specifies 16 and reality agrees. See datasheet (DS60001290D-page 243 figure 23-1)

CANopenNode, 2016-07-06:
It seems, that only device PIC32MX530 has 16 filters. PIC32MX534 seems to have 32 filters, as all other devices.
I fixed the CO_driver.c file. It should automatically detect PIC32MX530 device. If your device accepts more than 16 different CAN messages, CAN hardware filters will not be used and each received CAN message will be tested by software. There should not be any performance issues.
Please test the code.

Regards,
Janez

dmarcosgon, 2016-07-06:
It affects all the PIC's from the same family. PIC32MX5XX is divided in two subfamilies, those that relate to PIC32MX3XX which go up to 80 MHz and don't have PPS and those that relate to PIC32MX2XX, which go up to 50 MHz and have PPS. They should have called these new PIC32MX5XX differently...

This affects, at least, PIC32MX530F128, PIC32MX550F256 and PIC32MX570F512 which are the ones that have CAN in the subfamily.

CANopenNode, 2016-07-07:
Thank you. I missed the table in DS60001290D, page two.
PIC32MZ devices seems to all have 32 rx filters.

Main changes in CO_driver.c are now:
```
#ifdef __PIC32MX
#if (__PIC32_FEATURE_SET__ == 530) || (__PIC32_FEATURE_SET__ == 550) || (__PIC32_FEATURE_SET__ == 570)
    #define NO_CAN_RXF  16
#endif
#endif
#ifndef NO_CAN_RXF
    #define NO_CAN_RXF  32
#endif

....

    CANmodule->useCANrxFilters = (rxSize <= NO_CAN_RXF) ? true : false;
```

closed 2016-07-07 by dmarcosgon