CANopenNode/CANopenNode #11: Dual CAN interfaces

jhaws1982, 2016-07-26:
My application requires that I independently control two motors via two independent CAN buses (latency concerns on single bus).  I'm using Linux and socketCAN.

I'm looking at the CANopenNode library and it appears that it provides a nice implementation, however it also appears that I can only use a single CAN interface per application.  Is this correct?  I am guessing I would need to modify the library to support multiple interfaces?  Or is there a cleaner way to approach this?

FYI - I searched the discussion forums on SourceForge and found a thread from 2014 (https://sourceforge.net/p/canopennode/discussion/387151/thread/04054e35/?limit=25#37dc).  Is this still true?  Best approach would be to modify and implement multiple COO globals?  If so, is this something that may be desirable for the masses, thus would require me submitting a pull request?

Thanks!

CANopenNode, 2016-07-29:
It is possible to implement different combinations of CANopen devices and CAN channels. Stack is object-oriented. CANopen.c file must be edited.

The easiest approach may be to use one CANopen device (with one object dictionary with objects for multiple motors). PDOs may then be split into multiple CAN channels.

Please note, Linux is currently not best for hard real-time.

Some example for multiple CANopen could be interesting for someone. So some link is very welcome. But if solution is very specific or if it adds much complexity into the code, I wouldn’t like to include it into the stack. Otherwise I will accept a pull request.

jhaws1982, 2016-07-29:
> It is possible to implement different combinations of CANopen devices
> and CAN channels. Stack is object-oriented. CANopen.c file must be
> edited.
>
> The easiest approach may be to use one CANopen [...]

The approach I took was to eliminate the global variables altogether.
The user can now create a CO_t structure of their own and have as many
as they would like.  A pointer to this is passed into CO_init() and
everywhere else in CANopen.c.

Is there a reason for going the global variable route?  I feel that
eliminating the global variable makes the code more understandable and
flexible, at least from my perspective.  However, my experience is
fairly limited in the CAN world - maybe there is some other common
application where having globals simplifies things greatly.
> 
> Please note, Linux is currently not best for hard real-time.

Right.  Our application doesn't have hard real-time requirements.
> 
> Some example for multiple CANopen could be interesting for someone. So
> some link is very welcome. But if solution is very specific or if it
> adds much complexity into the code, I wouldn’t like to include it into
> the stack. Otherwise I will accept a pull request.

As I mentioned, I've got an implementation ready to test and I hope to
do so over the next few days.  If everything works I'll submit a pull
request so you can take a look at the changes.  Whether or not you
accept it is fine either way.  I've got a fork that I can pull from with
my changes.

Thanks!
> 
>

CANopenNode, 2016-07-29:
Hi,

Globals are sometimes nice for debugging on some microcontrollers. However, it is possible to have separate CANopenSimple.c, which uses only globals and CANopen.c without them.

Object dictionary is now implemented with globals only. I plan to change that in future. For now there is a workaround in OD editor. If you use a underscore in the beginning of project name, OD editor can generate globals prefixed with different names.

wiesniak1, 2016-10-24:
Hello.

I would be very interested in such change.  It is actually mandatory in larger scale solutions.
My kindly question is if you have a plans to integrate this change ? That would be great.

Best Regards
Greg.

closed 2017-02-02 by CANopenNode