CANopenNode/CANopenNode #12: Example files appear to use uninitialized pointer

RenardGraham, 2016-08-24:
Hello,
I am using your CANOpennode software on a SAMA5D2 processor for a project of ours.  
I got everything to make and was using the provided example main to just test that things were working before moving to implementing the driver interface when I encounted a problem.  
At the beginning of main.c  there are these lines:
`        /* disable CAN and CAN interrupts */
		printf("Disable CAN and interrupts\n\r");
		CO->CANmodule[0]->CANnormal = false;`

seems straight forward, but on my eval board the program hangs and won't continue after this statement. I get the output from the printf, and then nothing after that.
From what I can tell CO has been created, but it is set to NULL.
It only gets initialised after this call:
`        /* initialize CANopen */
		printf("Calling CO_init\n\r");
        err = CO_init(0/* CAN module address */, 10/* NodeID */, 250 /* bit rate */);
`

I  have looked at the PIC example and it is the same, yet I would assume it runs.
I feel like using the pointer before it's initialized is incorrect, and it doesn't work on my board, but I don't see anyone else complaining.
What am I missing here.

Thank you.

CANopenNode, 2016-08-31:
It seems, there is a mistake in mainline for PIC and SAM. You should erase that line.

That line is already included inside CO_init().

In main_PIC32.c, which is included in CANopenStack, that line is not present.

Examples for other PICs run, however, I will repair them.

RenardGraham, 2016-08-31:
I think it is also in the main.c generic as well.
Thanks, I wasn't sure it was a mistake, but taking it out fixed my implementation, and I then later noticed it in CO_init() as well, so I figured it was an oversight.

closed 2016-08-31 by RenardGraham