CANopenNode/CANopenNode #13: Pic18

komrisystems, 2016-09-29:
Are Microchip PIC 18's supported?

CANopenNode, 2016-10-10:
No, CANopen is hard to implement on 8-bit microcontrollers.
However, very old version of CANopenNode supported PIC18, see:
https://sourceforge.net/projects/canopennode/files/canopennode/

komrisystems, 2016-11-09:
Hello,

Thanks for the quick reply. 

I am working with a 32 bit micro-controller and it is my first time working with a CAN Open Stack. Do you have a wiring diagram for a PIC 32 and code example to get up and running quickly?

Sincerely

Karim

> 
> No, CANopen is hard to implement on 8-bit microcontrollers.
> However, very old version of CANopenNode supported PIC18, see:
> https://sourceforge.net/projects/canopennode/files/canopennode/ <https://sourceforge.net/projects/canopennode/files/canopennode/>

—
Karim Omri, M.A.Sc., P.Eng.
President
Komri Systems Incorporated

web: www.komrisystems.com <http://www.komrisystems.com/>
tel: (705) 618-2716

CANopenNode, 2016-11-09:
Please see here: https://github.com/CANopenNode/CANopenPIC

"Program is tested on Explorer16 board from Microchip, devices PIC32MX795F512L and dsPIC33FJ256GP710. CAN transciever chip is soldered to the Explorer16 board."

Wiring diagram for Explorer16 board is accessible from microchip.com.

komrisystems, 2016-11-15:
Are you using the PigTail Kit as well? Where is the CAN Transceiver chip solderer?

Thanks for the support,

Karim

> Please see here: https://github.com/CANopenNode/CANopenPIC <https://github.com/CANopenNode/CANopenPIC>
> "Program is tested on Explorer16 board from Microchip, devices PIC32MX795F512L and [...]

—
Karim Omri, M.A.Sc., P.Eng.
President
Komri Systems Incorporated

web: www.komrisystems.com <http://www.komrisystems.com/>
tel: (705) 618-2716

closed 2017-02-02 by CANopenNode