CANopenNode/CANopenNode #14: Can't write to subindex zero of register 0x1003 Pre-defined error field

Drllap, 2016-10-14:
It is my understanding that this should be possible, and doing it should reset the error counter.

I also found [this](https://github.com/CANopenNode/CANopenNode/blob/5df11b9d749b3802267f4c6d50259eb427c6139a/stack/CO_Emergency.c#L81) piece of code that suggests this has been thought of at some point.

If I try to do this write, the function above never gets executed, instead CO_SDO_AB_READONLY error is thrown [here](https://github.com/CANopenNode/CANopenNode/blob/5df11b9d749b3802267f4c6d50259eb427c6139a/stack/CO_SDO.c#L816)

This happens even though OD entry for this register in initialised like [this](https://github.com/CANopenNode/CANopenNode/blob/5df11b9d749b3802267f4c6d50259eb427c6139a/example/CO_OD.c#L306) with attribute of 0x8E and never changes. (Note that CO_ODA_READABLE == 0x04)

What happens is that when the object for this registered is fetched from the OD the attribute member is populated with [this](https://github.com/CANopenNode/CANopenNode/blob/5df11b9d749b3802267f4c6d50259eb427c6139a/stack/CO_SDO.c#L485) function that makes sure that subindex zero is never written to.

CANopenNode, 2016-10-21:
I'm currently short with my time. I have to update and test index 0x1003 more precisely. I plan to do this in ten days.

CANopenNode, 2017-02-03:
There was a problem with suggested fix, as described in comment to #15.

However, it is fixed now.
Thanks

closed 2017-02-03 by CANopenNode