CANopenNode/CANopenNode #18: sdo callbacks

IzidorM, 2016-12-12:
Hi,
I want to trigger some events via canopen sdo messages, like writting to index 0x2000 should start some measurement. As I can see, there is no callback functionality per OD entry, so I wonder what is the recomended way to figure out which OD entry has changed and call right handler?

jensnielsen, 2016-12-12:
Hi

Check out CO_OD_configure()

You can register a callback function for a specific index, which seems to be exactly what you want. There are examples throughout the code for how to use it.

//Jens

IzidorM, 2016-12-12:
Hi,
thanks for the fast replay. That is exactly what I need :) I was searching for this feature in the OD files, didnt expect that there is an extension of the OD implemented in the SDO files.

closed 2016-12-12 by IzidorM

Simon-Thome, 2021-04-29:
It seems the function CO_OD_configure() is not available in CANopenNode V4, or I am I missing something?

CANopenNode, 2021-04-29:
In V4 there is completely new Object Dictionary interface. See example in https://github.com/CANopenNode/CANopenSocket.

Simon-Thome, 2021-04-30:
So if I understand correctly. To create a callback function on an SDO write, using OD_extension_init a custom write function can be created which can call a callback function and OD_writeOriginal can then be used to perform the actual write. Is this the best approach?