CANopenNode/CANopenNode #20: Rewriting the ODE

WillyKaze, 2017-01-10:
Using the ODE based on XUL files didn't work for me after many hours of research.

So I started to develop a new ODE written in python3.
Hopefully this will make scripting really easy.

You can see my working copy at [ExMachina repository](https://github.com/exmachina-dev/CANopenNode/tree/feature/new_ODE)

I'll make a PR once it's usable. Right now only the data structure and the configuration file is working.

Comments and remarks are welcome.

Best regards,
Ben

CANopenNode, 2017-02-03:
Hi,

There are plans to renew Object dictionary part of CANopenNode.

I planed to use javascript and json for databases (because I'm more skilled with it). GUI will not be necessary.

Your python solution seems interesting too. Can you give a short instructions, how to use it.

I'm currently using ODE based on XUL. It works, but it is old and complicated.

Here are short instructions, how I ran it on 64-bit Ubuntu:
 - https://ftp.mozilla.org/pub/firefox/releases/16.0.2/linux-x86_64/en-US/
 - Extract to firefox-16.0.2 directory somewhere.
 - Version 3.6 does not work. On 64 bit system works x86_64.
 - `firefox -P`
 - Create new profile `firefox-16.0.2`, then exit.
 - `./firefox-16.0.2/firefox -no-remote -P firefox-16.0.2`
 - Edit -> Preferences -> Advanced -> Update -> Never check for updates
 - XUL:
   - https://addons.mozilla.org/en-US/firefox/addon/remote-xul-manager/versions/?page=1#version-1.1.3.1-signed
   - Install version 1.1.3
   - Tools -> Web Developer -> Remote XUL Manager -> Add... -> `'<file>'`

 - See also:
   - http://ubuntuforums.org/showthread.php?t=1907045
   - http://blog.oldversion.com/how-to-install-and-run-multiple-versions-of-firefox/

CANopenNode, 2017-02-04:
There are some other efforts for OD editor:

Nice GUI: https://github.com/robincornelius/libedssharp

A Python tool for use with old OD Editor: https://sourceforge.net/p/canopennode/discussion/387151/thread/14e56935/

robincornelius, 2017-02-04:
I'm the author behind the libedssharp app listed on the last post. My tool does currently support direct export to the CO_OD.c and CO_OD.h files in their current format as well as reading/writing the xml format used by the javascript tool and of cause reading/writing eds files. It is in C# so should run on windows as well as mono for mac/linux. There are two parts to it a library and a GUI. The library is capable of the most of the work with out the GUI so could be used as a command line tool (in theory but more work may be needed as i've never actually tested this myself yet).

And if CanOpenNode changes its OD format i will try to also support that

WillyKaze, 2017-02-06:
Ho, nice work !

I'm not sure I want to add yet another alternative. My approach is based on a config file in INI format. This is meant to be editable by human (example [here](https://github.com/exmachina-dev/WDY-firmware/blob/master/OD/mbed.conf)). My first regret is that the configuration file is quite big (~2500 lines).

I'll try @robincornelius 's tool but it seems he's got everything worked out.

c0d3runn3r, 2017-03-15:
I am writing a CANOpenNode driver for Arduino Teensy3 (Kinetis K20). Currently the autogenerated CO_OD.h has that
```   typedef domain_t     DOMAIN;```
Which is a small problem because DOMAIN is a reserved word on this platform (arm math.h has ```#define DOMAIN 1```).  It would be nice if the next generation of OD editor used a different keyword (perhaps CO_DOMAIN).

CANopenNode, 2017-03-16:
A will do that.

c0d3runn3r, 2017-03-28:
I downloaded https://github.com/robincornelius/libedssharp and spent some time with it.  It's a little rough around the edges, but in general I am EXTREMELY HAPPY with it.  Not least because it exports .c/.h files suitable for import into CANopenNode.  Very painless.  I suggest considering them as the "official" ODE for CANopenNode.

martinwag, 2017-05-10:
> Which is a small problem because DOMAIN is a reserved word on this platform (arm math.h has #define DOMAIN 1). It would be nice if the next generation of OD editor used a different keyword (perhaps 

This is not specific to arduino. This is also a problem on math.h on PC Linux.

kenshail, 2017-11-25:
Using the excellent EDSeditor V0.5.2 or 6 beta I have one problem: I can not get either version to set CO_NO_NMT_MASTER to 1. I am using it with the .eds file that comes with CANopenSocket. Any suggestions?

jiangj001, 2018-11-01:
@c0d3runn3r I'm looking for a can open lib working for Teensy 3.2, did you eventually finish that project? Is that possible for you to share yours, please?

c0d3runn3r, 2018-11-01:
Yes I did, almost two years ago.  @CANopenNode still hasn't merged my pull request though.  You can find it here: https://github.com/CANopenNode/CANopenNode/pull/28

CANopenNode, 2020-01-10:
I recommended https://github.com/robincornelius/libedssharp for OD editor in README.

lib for Teensy 3.2 is listed in wiki.

I will try to rewrite OD part of CANopenNode in next few months.

closed 2020-01-10 by CANopenNode