CANopenNode/CANopenNode #21: reading DOMAIN via SDO > SDOBufferSize fails

robincornelius, 2017-02-08:
I may be not understanding the way this is suppose to work but if i have a domain variable and i want to transfer a bigger buffer than CO_SDO_BUFFER_SIZE then it fails with an SDO abort of CO_SDO_AB_DEVICE_INCOMPAT

I have defined a OD access function via CO_OD_configure()  and then i am doing the following in my function :-

    if(ODF_arg->reading)
    {
    
                ODF_arg->dataLengthTotal = com1_buffer.count;            
                ODF_arg->dataLength = com1_buffer.count;
                ODF_arg->data = com1_buffer.buffer;
    }

So i have updated the data pointer to point at my buffer and set the sizes

if i set the ODF_arg->dataLength > CO_SDO_BUFFER_SIZE then the SDO transfer fails. If i remove the check on line 641 of CO_SDO.c which says :-

    /* dataLength (upadted by pODFunc) must be inside limits */
            if((SDO->ODF_arg.dataLength == 0U) || (SDO->ODF_arg.dataLength > SDOBufferSize)){
                return CO_SDO_AB_DEVICE_INCOMPAT;     /* general internal incompatibility in the device */
            }

then it works perfectly and i get all bytes transferred by SDO (in my case a segmented sdo transfer)

Reading the comments on CO_SDO.h around line 323 it states
     * Size must be at least equal to size of largest variable in @ref CO_SDO_objectDictionary.
     * If data type is domain, data length is not limited to SDO buffer size.

So that comment would suggest that the above check should not be present for DOMAIN type variables?

Have i made a mistake with my implementation or is this a bug?

Thanks!

CANopenNode, 2017-02-09:
Please check function in line 200 from here: https://github.com/CANopenNode/CANopenPIC/blob/master/appl_PIC32_explorer16_IO.c

Domain should work, but I haven't used it last time.

ODF_testDomain() is called multiple times from SDO, it depends how many data is there.

robincornelius, 2017-02-09:
I still believe the line i highlighted is at fault. I have applied the following patch [0001-Ensure-its-not-a-DOMAIN-transfer-by-checking-ODdata-.txt](https://github.com/CANopenNode/CANopenNode/files/763579/0001-Ensure-its-not-a-DOMAIN-transfer-by-checking-ODdata-.txt)

And my domain transfers are working correctly for both upload and download now.

robincornelius, 2017-02-09:
I've spent a bit more time going through the code and it looks like although i had things working it was not in an intended way. With my patch it is possible to update ODF_arg->data pointer to a user buffer and just set the totallength and things appeared to work. What does not work is if you have a zero length buffer for the domain. in this case it completes an SDO transfer with 4 data bytes all set to zero which is clearly wrong.

I've now reimplemented things as per the PIC32 explorer16 example and everything appears to function without the patch.

Thanks for the pointer in the correct direction!

closed 2017-02-09 by robincornelius