CANopenNode/CANopenNode #24: ODL_errorStatusBits_stringLength makes custom object mandatory

trojanobelix, 2017-02-21:
Not quite sure this an issue, but the ODL_errorStatusBits_stringLength must be at least 10, that makes the custom object 0x2100  mandatory. And this is not according DS301.

```
/* Verify features from CO_OD *************************************************/
    /* generate error, if features are not correctly configured for this project */
    #if        CO_NO_NMT_MASTER                           >  1     \
            || CO_NO_SYNC                                 != 1     \
            || CO_NO_EMERGENCY                            != 1     \
            || CO_NO_SDO_SERVER                           == 0     \
            || (CO_NO_SDO_CLIENT != 0 && CO_NO_SDO_CLIENT != 1)    \
            || (CO_NO_RPDO < 1 || CO_NO_RPDO > 0x200)              \
            || (CO_NO_TPDO < 1 || CO_NO_TPDO > 0x200)              \
            || ODL_consumerHeartbeatTime_arrayLength      == 0     \
            **|| ODL_errorStatusBits_stringLength           < 10**
        #error Features from CO_OD.h file are not corectly configured for this project!
    #endif
```

CANopenNode, 2017-02-24:
You are probably right.
Now I'm making new OD editor and OD interface and this problem will be solved also.

If it is necessary, you can also edit CANopen.c a little - for example make static array for errorStatusBits and remove OD_errorStatusBits from Object dictionary.

trojanobelix, 2017-02-27:
2100h is not in use on my project. So for me it is not a problem. I just want you to have a quick view on it.

closed 2017-02-27 by trojanobelix