CANopenNode/CANopenNode #26: problem with heartbeat

franck48, 2017-03-22:
when the master stop to send heartbeat to canopennode 'node'
the node go to pre-op-> this is ok
but the node was locked, impossible to return to operational state.
canopen node dosn't accept the nmt message

CANopenNode, 2017-03-22:
This must be problem with driver. On PICs and Linux are not happening such things (as I know). But I can not guaranty for other microcontrollers. Which one do you use?

franck48, 2017-03-23:
i use a microchip 16bit microcontroler.

to be more clear.

on the power up, the master send with SDO messages parameters to canopen node so that my board have to survey the master heartbeat.

then,
i go to operational and all is ok

then,

the master stop to send his heartbeat,

immediately my board goes to preop

but,

when the master or with a probe i send the nmt message to return to operational canopen node don't care of this message,
the only way to restart the node is to reboot the node.

i have do so reseah inside the source code of the stack, i found probably the reaon why the nmt message is discarded :

inside the file CO_NMT_heratbeat.c,  there is this function ,
INTEGER16 CO_NMT_receive(void *object, CO_CANrxMsg_t *msg)

look this,
 switch(command)
{
         case CO_NMT_ENTER_OPERATIONAL:      if(!(*NMT->EMpr->errorRegister))
                                                                                               NMT->operatingState = CO_NMT_OPERATIONAL;    break;

before to go to operational, canopennode check the error register.

but even if i comment this check /*  if(!(*NMT->EMpr->errorRegister)) */ it is not sufficient.

so i decide to try to clean the master heartbeat default like this
CO_errorReset(CO->HBcons->EM, ERROR_HEARTBEAT_CONSUMER, 0);

but the problem continue, i have 20 canopen messages sended before the clean succeed and a 21th message
with 'no-error'

so i continue and decide to call in a lopp this function until it succeed
while( CO_errorReset(CO->HBcons->EM, ERROR_HEARTBEAT_CONSUMER, 0) != 1 )
 { ; }

now it's better but the error continue to appear.

i decide to restart the heartbeat message from the master and only after this to send the nmt message to go to OPerational.
but this doesn't work.

finally i found this code to solve but i'm not satisfied.

CO_HBconsNode_t *monitoredNode = &CO->HBcons->monitoredNodes[0];
            if( CO_isError(CO->HBcons->EM, 0x1B, 0))
            {
                if(CO->NMT->operatingState == CO_NMT_PRE_OPERATIONAL )
                {
                    CO->NMT->EMpr->errorRegister = 0;
                    OD_errorRegister = 0;
                    CO->NMT->EMpr->reportBuffer->errorStatusBits = 0;
                    while( CO_errorReset(CO->HBcons->EM, ERROR_HEARTBEAT_CONSUMER, 0) != 1 )
                    {
                        CO->HBcons->EM->errorStatusBits = 0;
                        CO->HBcons->EM->msgBufferFull = 0;
                    }
                }
            }

franck48, 2017-03-23:
> As I understand, your device is functioning, it just don't go into operational.
>
> Here are some general rules, which you probably already know:
> If error register is different than zero, device won't go into operational. You can change this by Error behavior (1029).
> You can check errors:
>
>   *   Error register (0x1001),
>   *   Error status bits(0x2100) - you can see, which error is setting Error register
>   *   Emergency messages
>
> Currently CANopenNode doesn't provide recovery after Heartbeat consumer error. It is quite serious error in CAN communication. You should set heartbeat consumer large enough, that this does not happen. You can also disable monitoring of the specific node, if you plan to shutdown it.
>
> If you really need automatic recovery after heartbeat consumer error, there is probably some more work to do. You should not directly manipulate EM variables. (CO->NMT->EMpr->errorRegister is calculated somewhere else, CO->HBcons->EM->errorStatusBits is a pointer, etc)
>
> If Heartbeat consumer error happens anyway is most elegant to just reset communication.

yes you are right, it's a big default.

instead of trying to modify the stack, i finally prefer to follow the canopen node developer's rule :

--> in case of heartbeat master fault, the master must send nmt message to reset the node.

thank you for your help

and to the 'man' behind this stack for his free and wonderfull job

CANopenNode, 2017-03-24:
It is a goal for CANopen devices to just keep working all the time. It is quite possible, if device, power supply and wiring is good.
Otherwise also resetting communication is not a big deal. The rest of device can run uninterrupted.

I think, I can close the issue.

closed 2017-03-24 by CANopenNode