CANopenNode/CANopenNode #27: Added K20 driver

c0d3runn3r, 2017-03-23:
My company is writing ARM code for the Kinetis K20.  This is the platform also used by PJRC's Teensyduino.  

I've written a driver for the K20 and so far it seems to be working great.  I've commited the driver to a local branch, if you want to give me push access I'll add it and create a pull request so you can look it over.

Disclaimer: I've commented the code so it's reasonably legible, but I haven't cleaned it up to ensure it meets all the coding style guidelines.  I've tested SDO, HB, and HB consumer but not PDO.

CANopenNode, 2017-03-23:
Hello,

If you fork the original CANopenNode, you will have the possibility to pull request. This is most practical approach.

Please see also https://github.com/CANopenNode/CANopenNode#note-for-contributors

c0d3runn3r, 2017-03-23:
https://github.com/CANopenNode/CANopenNode/pull/28

CANopenNode, 2017-03-26:
Here are some ideas to split driver files for all different microcontrollers to separate git repositories. I'm currently using only PICs and Linux and can't maintain the other drivers. Driver authors should have control and maintainability over their code. (There are no plans to change driver interface in CANopenNode).

User will have to include two git submodules into project: CANopenNode + driver.

I will try to maintain links to available drivers in CANopenNode.

Some more discussion is in #25

c0d3runn3r, 2017-04-25:
OK, so you want me to create my own repo with the driver files and then give you the URL?  (or pull request to update the CANopenNode README.md with the K20 driver URL?)

CANopenNode, 2017-04-29:
Sorry, I currently don't have much time for CANopen.

For now, I'm just leaving this issue open.
I think, I will make some kind of public editable wiki.

martinwag, 2019-10-08:
new drivers are handled in #108

closed 2019-10-08 by martinwag