CANopenNode/CANopenNode #29: IO Demo code Documentation?

mgiaco, 2017-03-31:
Hello,

I am getting my sample for the STM32F103 Olimex P103 demo more and more to work. So simple read and write seems to work.  Now I do not really know how to add the interaction with the input and output´s. Is it right that the demo should support 8 digital in, 8 digital out, and 8 analog in, and 8 analog out?  

I also study the STM32 example which i found in the old SVN repo i don't see any read and write of the outputs or inputs whether analog or digital. 

I do not have any PIC but I tried to study the CANopenNodePIC demo also. Does this demo also use the main from the CANopenNode Stack example ... so is it also an IO demo using the IO.eds file from the stack reop? It isn´t really not easy for someone which is quite new to canopen. 

So can someone help me: Which file is the right one for the IO example? How does the interaction with the IO´s work? And where should I read and write the output?

many thanks.
cheers
mathias

CANopenNode, 2017-04-07:
I understand, it is hard to start with CANopen and CANopenNode. I can hardly find my time to make CANopenNode and documentation better, it usually goes together with my projects. Currently I'm working on improvenent in Object dictionary interface.

I can't help you with STM (currently), but integration of inputs and outputs is quite simple. At the end of CO_OD.h you find variables, which you can just use.

For example from https://github.com/CANopenNode/CANopenPIC/blob/master/appl_PIC32_explorer16_IO.c 
´´´
    /* Read RPDO and show it on LEDS on Explorer16. */
    leds = OD_writeOutput8Bit[0];
    LATAbits.LATA2 = (leds&0x04) ? 1 : 0;
´´´
(Actually part of CANopen RPDO is configured to write into OD_writeOutput8Bit[0]. This is configured according to CANopen standards. OD_writeOutput8Bit is part of Object dictionary. LATA... switches the led.)

mgiaco, 2017-04-07:
Hello, 
I already found out how that works i think. But I will ask back if I have any problem. One of the most important step to this stack is to add samples for STM32 which work out of the box for some board. I will work on that. But general I think it is really confusion if the main of an example is in the stack driver for that cpu. That was why I first do not found out how that works together.
I thought if I checkout the CANopenPIC the complete example is inside this repo but the main is in the stack :-)

So why not also separate that from the stack.

CANopenNode, 2017-04-08:
I will separate all driver files from the stack. I will maintain only microcontrollers, which I can run. There will be links to other implementations

.

closed 2017-04-08 by CANopenNode