CANopenNode/CANopenNode #30: Problem with heartbeat

ValAGS, 2017-04-06:
Hello,

During the development of my application, I have found a problem with the emission of the heartbeat when I choose to reset node.
In function CO_NMT_process, a negative value can appear and the heartbeat will be emit 3 or 4 times every ms.
So, I made a correction that I share here : 

/* Heartbeat producer message & Bootup message */
    if((HBtime && NMT->HBproducerTimer >= HBtime) || NMT->operatingState == CO_NMT_INITIALIZING){

        NMT->HBproducerTimer = NMT->HBproducerTimer - HBtime;

        NMT->HB_TXbuff->data[0] = NMT->operatingState;
        CO_CANsend(NMT->HB_CANdev, NMT->HB_TXbuff);

        if(NMT->operatingState == CO_NMT_INITIALIZING){
						if(HBtime > NMT->firstHBTime) 
							NMT->HBproducerTimer = HBtime - NMT->firstHBTime;
	// ! CORRECTION ! //
					        else
							NMT->HBproducerTimer = 0;
        // ! CORRECTION ! //
            if((NMTstartup & 0x04) == 0) NMT->operatingState = CO_NMT_OPERATIONAL;
            else                         NMT->operatingState = CO_NMT_PRE_OPERATIONAL;
        }
    }

Best regards,

Valentin

jensnielsen, 2017-04-06:
Hi

Didn't I fix this already 1½ year ago in  ba1265138a55e07e1e22f37a4b71ee459d00d190?

Or am I missing something? Are you using an old version?

Best regards
Jens

ValAGS, 2017-04-06:
Hi

You're right I use an old version (July 2014).
By the way, is there many corrections since 3 years in the functionment of the canopennode stack ?

Valentin

CANopenNode, 2017-04-08:
I'm not sure. If you have directory "\stack\drvTemplate", then there are no fundamental changes. There was some clean-up, rearrangement in github and bugfixes.

Janez

closed 2017-04-08 by CANopenNode