CANopenNode/CANopenNode #47: Adding LSS functionality

martinwag, 2017-06-26:
As mentioned here (https://sourceforge.net/p/canopennode/discussion/387151/thread/397f3912/) I'm in the process of implementing LSS.
I've pushed a first version to my local repo in lss branch (https://github.com/martinwag/CANopenNode/tree/add-lss). Feel free to have a look, but be aware that this is still experimental. If you find errors or have other hints, please let me know :-)

Current limitations:
- needs #define CO_NO_LSS_SERVER 1 in CO_OD.h. This will need a change in the OD editor.
- no Fastscan support yet
- bitrate setting API untested. I've implemented this for the sake of completeness, I don't need it in my application.
- https://github.com/CANopenNode/CANopenNode/issues/46 (driver stuff)
- changed some of the other stack files, tried my best to remain compatible with standard non LSS CANopenNode
- there is a bit of a mess about server/slave and master/client. But this is a mess in the other parts of the stack too :-(

CANopenNode, 2017-07-04:
This is very serious contribution to CANopenNode. I will review driver interface and add LSS to the stack.

martinwag, 2017-07-04:
Please do not merge the stuff yet. I'm still in the middle of implementing myself :-).

martinwag, 2017-07-13:
I've pushed LSS master and LSS fastscan, so LSS should be complete now. This is the updated list:

Current features
- LSS slave
- LSS master
- Fastscan support

Current limitations:

- needs testing.
- needs #define CO_NO_LSS_SERVER 1 or  #define CO_NO_LSS_CLIENT 1 in CO_OD.h. This will need a change in the OD editor. Only one can be active at a time.
- bitrate setting API untested. I've implemented this for the sake of completeness, I don't need it in my application.
- #46 (driver stuff)
- changed some of the other stack files, tried my best to remain compatible with standard non LSS CANopenNode
- there is a bit of a mess about server/slave and master/client. But this is a mess in the other parts of the stack too :-(

iurly, 2017-11-21:
Hi,
I recently started checking out this piece of work and I gotta say it works great and it's really well written and documented!
It just took me a little while to figure out the whole LSS state machine (i.e. come up with some meaningful test cases) but that's probably my fault.

I have one question though: is there a particular reason why the 3 callbacks are called in interrupt context? This forces me to add an extra level of queuing, whereas it would've probably been easier to just store a flag and have the callbacks invoked from CO_LSSslave_process() at the next call.
But perhaps I'm missing something.
Thanks again!

martinwag, 2017-11-21:
Yes, there is a simple reason :-). It was easier for me. I have an RTOS, so outside of CANopenNode I can just use ready-made functions for synchronisation.

The LSS state machine is quite tricky to get right, especially if you want to have LSS fastscan. Follow the examples in the specification really carefully.

iurly, 2017-11-21:
You're right that state machine is tricky. What I'm not quite following is the logic of the Node-ID assignment. By looking at the example code within the comments of CO_LSSslave.h, it appears like setting it for the first time on an unconfigured node just activates it immediately (so you should immediately get NMT running), whereas if you do it a second time, you shall first issue a RESET_COMM in order to activate it. Is my understanding correct? Was this done on purpose?

Quite frankly, I'm not quite happy with the extra start_canopen() loop for an unconfigured node. Even when unconfigured, a bare metal implementation should also do some other stuff as well (e.g. poll other communication interfaces, drive LEDs, re-arm the watchdog and so on). So I'd rather have everything in a single main loop, which initializes NMT only when a valid NodeID is found.

martinwag, 2017-11-22:
The start_canopen() function has two use cases.
- Start after boot
- Start after RESET_COMM

After boot, you either get a valid node ID from NVM or fastscan mode is started (the loop). In this mode, only the LSS slave is active. This implements the functionallity required in CiA305 Figure 48.
On RESET_COMM, the same functionallity is used, except that a node ID is already known, so the loop is always skipped. Which node ID is used in this case is explained in Figure 4.

After start-up LSS is serviced like NMT.

I'm not really happy with the startup code either. I think you have two options here:
- Split the two start-up modes into two functions
- Implement a start up state machine

I've tried to keep the example code short and in a -more or less easy- to understand order...

Ref: CiA® 305 Draft Standard Proposal Version: 3.0.0 08 May 2013

gotocoffee1, 2018-04-25:
Hey @martinwag ,
i am trying your LSS Fork and it looks very promising. Would be cool to see that in the main repo.
I have one thing that my middleage compiler doesn't support, the unnamed struct [CO_LSS_address_t](https://github.com/martinwag/CANopenNode/blob/add-lss/stack/CO_LSS.h#L187).
Could we consider something like:

    typedef union {
        uint32_t addr[4];
        struct {
            uint32_t vendorID;
            uint32_t productCode;
            uint32_t revisionNumber;
            uint32_t serialNumber;
        } identity;
    } CO_LSS_address_t;

best regards.

martinwag, 2018-04-26:
just create a pull request onto my branch. We're still in testing, when this is done I will create a request for the main repo.

martinwag, 2019-10-08:
implemented in 8b01062eb6568721304164db9f0fea1fde546e76

closed 2019-10-08 by martinwag