CANopenNode/CANopenNode #48: thread/interrupt signaling

martinwag, 2017-07-05:
While looking at error https://github.com/CANopenNode/CANopenNode/issues/39, I came across receive message signaling. This is currently done by copying the message and then setting a flag in one thread/interrupt and evaluation in another thread/main loop. This might work on some embedded compilers (like the free microchip ones), but can break when enabling compiler optimization. As soon as you do that, your compiler is allowed to shuffle around memory copy operations as it likes to do. So you have no guarantee that `xxx->CANrxNew = true;` is actually executed last.

I'm thinking about the 'best way' to fix this. I can see some options that might be viable:
- Add memory barrier and set the flag as volatile variable to ensure copy and evaluation order. With GCC this is `__sync_synchronize()`.  I have no idea how (and if) to add the memory barrier with other compilers.
- move this signaling to stack specific queueing functions (like it is done with e.g. memcpy), where the user can decide it the current behaviour is acceptable, or do something different like use the OSes queuing functions.

CANopenNode, 2017-07-05:
I wasn't aware that compiler can shuffle memory copy operations. This is quite a serious issue with CANrxNew flags...

martinwag, 2017-07-05:
I've had a little look around and found this, which I think explains the problem in short:
https://stackoverflow.com/questions/22106843/gccs-reordering-of-read-write-instructions
- compiler can reorder access
- cpu can reorder access 
- memory might differ in caches on multicore cpu
- volatile isn't a guaranteed way to disable re-ordering, it only forces that a variable is always read from and written to ram.

The last three cases will most likely only be relevant on higher class systems with full flavour OSses (windows, linux), which is why almost any documentation for those recommends not to do inter thread sync with flags.

CANopenNode, 2017-07-06:
I was thinking a little about that problem.
I think, this dual-processing-of received-messages is good approach in CANopenNode. All received CAN messages are first shortly pre-processed by (fast) thread. Further processing can then be handled slowly.
This approach may not be so beneficial in Linux, because it anyway does not have hard real-time. But on microcontrollers, especially slower ones or those with small CAN receive buffer, may offer much better characteristics.

In (my) PIC32 driver CAN receive is implemented with highest priority interrupt. So, I suppose, shuffling  memory copy operations is not a problem in receive function, but may be a problem in functions, which processes the message. So things need to be reconsidered more carefully.
Solutions will probably have to be OS and microcontroller specific.

martinwag, 2017-09-01:
I've had a play with that in LSS, and came to the following solution.

```
#ifndef __GNUC__
/* this GCC builtin function ensures memory barriers. It might be needed when
 * the compiler or cpu re-orders memory access */
#define __sync_synchronize
#endif
#define IS_CANrxNew() (LSSmaster->CANrxNew)
#define SET_CANrxNew() __sync_synchronize(); LSSmaster->CANrxNew = true;
#define CLEAR_CANrxNew() __sync_synchronize(); LSSmaster->CANrxNew = false;
```

We could make this universal and move it to the driver, like the locking functions. I think it should then work like that:
- replace CANrxNew flags data type to void*. The user then can select in the function implementation if he wants to use this as a simple flag or pointer to OS syncronisation variable or something else
- replace all occurances to the flag with the corresponding get/set macro.

What do you think about that? The biggest problem I see with that is, that it has to be added to all drivers (including testing)...

martinwag, 2017-09-01:
I've added this, you can have a look here: https://github.com/martinwag/CANopenNode/tree/memory-sync . I've modified the template and linux driver.
Please do not pull this, it's just an experiment.

I've changed the flag to a void* type, so we can use this as reference/pointer for a OS lock as well.

martinwag, 2019-10-08:
This is integrated in 8b01062eb6568721304164db9f0fea1fde546e76

closed 2019-10-08 by martinwag