CANopenNode/CANopenNode #50: struct sCO_OD_ROM CO_OD_ROM - const?

martinwag, 2017-08-02:
Hi,

I can't find this in the docs - is there a reason that I shouldn't declare the ROM OD entries struct as const? This basically gives the compiler the option to not place this stuff inside RAM.

In the current implementation, I can't find a difference between ROM and RAM, both are loaded to ram, both are lost at reset.

robincornelius, 2017-08-02:
if you enable the USE_EEPROM define, the rom is infact loaded from the
lower part of the EEPROM (if checksum is valid) and is treated more like
many other can modules treat their eeprom so if you issue a save via
0x1010/0x01 with 'SAVE'  it will save the current "rom" values back to
eeprom. (I've never actually tried to save them back but i have updated my
firmware and been scratching my head to why the value used was different to
my code and this is the reason)

Values in CO_OD_EEPROM are updated and saved to eeprom as soon as they
change, so make sense for poweron counters etc were as values in CO_OD_ROM
should only save if you use the 0x1010/01

Robin

martinwag, 2017-08-02:
Thanks, found it in the eeprom driver example.

I've used it a bit different. In my use case ROM contains values that can't be changed, and EEPROM is saved using the 0x1010/0x1011 indexes. I think, I can life with that :-).

closed 2017-08-02 by martinwag

CANopenNode, 2017-08-05:
Originally (in my dsPIC30) CO_OD_ROM was declared as const and it was not placed into RAM, it was in flash memory. However, some variables from CO_OD_ROM are changeable from network side (for example PDO parameters). On parameter change (SDO write), function writes directly to flash. I didn't like this principle very much, so all those parameters are now in RAM. My drivers for Linux and PIC32 include ability to (re)store CO_OD_ROM variables with objects 1010/1011.

---
Plans with new Object Dictionary are, that structures like CO_OD_ROM won't be necessary any more. Variables will be for example part of objects (RPDO mapping can be part of CO_RPDO_t structure). Object dictionary will have pointers to variables. Object 1010 will scan over OD and save values from selected variables.