CANopenNode/CANopenNode #51: Simultaneous SDO Transactions

evilGitHub, 2017-08-09:
I would like to know if it is possible with this stack to to multiple simultaneous SDO transactions.

Like reading from two different nodes at once. As I understand I have to call CO_SDOclient_setup every time my target changes before using CO_SDOclientDownloadInitiate or CO_SDOclientUploadInitiate. Is it somehow possible (like creating two instances of CO_SDOclient_t) to do to simultaneous transactions?

CANopenNode, 2017-08-11:
I didn't try this, but it should work as you describe. (create two instances of CO_SDOclient_t inside CANopen.c, etc. Accessed nodes must be different.)

evilGitHub, 2017-08-15:
Ok as this seems not to be intented I dont want to try it. I just channeld all SDO communications through a seperate Thread now.

Besides: Thanks for this great work!

closed 2017-08-15 by evilGitHub