CANopenNode/CANopenNode #52: CANOpenNode + FreeRTOS

antony-sm, 2017-08-24:
Hi there!

 Sorry for my English)
 I've reconstructed this code to work under OS control, using queues, semaphores and other instruments. Want to push it here and make a public property. How can I do it?
 Please contact me: antony-sm@yandex.ru

martinwag, 2017-08-30:
I would be very interested in that. You can fork this repo and then push your code. This is explained [here](https://gist.github.com/Chaser324/ce0505fbed06b947d962). You don't have to create a pull request.

CANopenNode, 2017-08-30:
I agree with Martin.
I can't maintain code for all different hardware. So plans for CANopenNode are, that it should stay thin. Only the stack, which will be universal for all systems.

I suggest creating new git repository with (example) project for FreeRTOS. Then add CANopenNode as a git submodule into your project. Use all the stack files, but ignore the "driver" files (all subdirectories from CANopenNode/stack/ directory). Driver files for FreeRTOS should be the part of your project.

If there will be a need to change the files directly under the CANopenNode/stack/ directory, then do, as Martin suggested, fork the CANopenNode and then you can make a pull request to original CANopenNode.

There is a wiki (https://github.com/CANopenNode/CANopenNode/wiki), which is currently very empty, but anyone can write there a link with some description of his CANopenNode implementation.

antony-sm, 2017-08-30:
All's done. Enjoy

martinwag, 2017-08-31:
I've had a quick look at this, and I don't think this is a fork. I think this qualifies as an entiere new project for itselve :-).
- C++
- FreeRTOS
- Other folder structure
- UML Docs (unfortunately partly in russian)
- doxyfile seems to have been lost somewhere

antony-sm, 2017-08-31:
Sorry for russian docs :) The project initally was intended for russian robotics company) I've left them some time ago.
 Now I'm not sure that this project is usefull for that company without me)) And want to code not to die and be useful for everyone interested. I'll develop and debug it in my nowday "smart home" project

martinwag, 2019-10-08:
see https://github.com/CANopenNode/CANopenNode/pull/53

closed 2019-10-08 by martinwag