CANopenNode/CANopenNode #57: GPL License Questions

xoduddk123, 2017-09-12:
My working environment is Linux.
I tested CANopen using CANopenSocket's canopend. It worked well.
I created shared memory in canopend and tried to exchange PDO data with other programs.

I know that CANopenNode is a [GPL linking exception](https://github.com/CANopenNode/CANopenNode#gpl-linking-exception).
However, CANopenSocket is a generic [GPL v2.](https://github.com/CANopenNode/CANopenSocket#license)

I tested with [canopend](https://github.com/CANopenNode/CANopenSocket/tree/master/canopend) and tried to use shared memory, but gave up because it was GPL v2.
So I tried to use the GPL linking exception [CANopenNode/CANopenNode/example/main.c](https://github.com/CANopenNode/CANopenNode/blob/master/example/main.c).
After analyzing the source, I found that the [CANopenNode/CANopenSocket/canopend/src/main.c](https://github.com/CANopenNode/CANopenSocket/blob/master/canopend/src/main.c) and CANopenNode stacks were very similar.
Both CANopenNode/CANopenNode/example/main.c and  CANopenNode/CANopenSocket/canopend/src/main.c both use the stack of CANopenNode.

Both are thought of as a kind of daemon using CANopenNode.
This part confuses me.

1. CANopenNode is a GPL linking exception, and canopend is GPL v2 intentionally?
2. Can I refer to canopend's source when modifying CANopenNode/CANopenNode/example/main.c?

I am a newbie in connection with the GPL license.

I tried to find the mailing list but could not find it.
Is it a question to ask here?

I'm sorry I do not have enough English.

martinwag, 2017-09-25:
GPL genreally allows you to go from a less restrictive to a more restrictive license for your fork, but not the other way around. Restrictive in GPL terms means "open source-ness".
You can do LGPL -> GPL
You can do GPL Linking exception -> GPL without linking exception
You can do GPL V2 -> GPL V3

However, you can not do GPL -> LGPL without permission of the project maintainer.
You also can not do *GPL -> MIT ore something like that.

to your questions
1. I can't say for sure, but I assume that is the case.
2. If you want to create a commercial application, it doesn't really matter. You can't use either as "main" for your project as both have GPL license requiring you to open-source at least this one file. Take the stuff you need and pack it into a separate file(s), while having a look to clearly separate the CANopenNode stuff from your application.

xoduddk123, 2017-09-25:
Thank you.
It has helped a lot.
I will use the CANopenNode stack to clearly distinguish between CANopenNode and the application.

closed 2017-09-25 by xoduddk123