CANopenNode/CANopenNode #60: GPL License Questions #2 (socketCAN driver has no linking exception)

martinwag, 2017-09-25:
Hi Janez,

currently, CANopenNode is marked in the Readme.md as using GPL with linking exception. This is true for all sources, except the socketCAN driver. Those are GPLv2.
I would like to use this driver (and fix some bugs/add features to it). However GPL means that I can't integrate it into my application, forcing me to build a completely new one.

Can we change that to the license that is used for the driver template?

Thanks!

xoduddk123, 2017-09-26:
I posted a question on the [GPL License Questions](https://github.com/CANopenNode/CANopenNode/issues/57).
I also use socketCAN and I was surprised to see that it was only GPL v2.
I want socketCAN to be a GPL linking exception.

I'm sorry I do not have enough English. :)

Thanks!

martinwag, 2017-09-26:
I know, I'm the one who answered your question :-)

petmos, 2018-01-24:
Any news for this issue?
All sockcan files were commit on 10. Jan 2016 with " License changed to the GNU General Public License version 2 with the classpath exception."
But this exception wasn't included in these socketcan files.

martinwag, 2019-10-08:
Got around this by writing my own...

https://github.com/martinwag/CANopenNode/tree/neuberger-master/stack/neuberger-socketCAN

closed 2019-10-08 by martinwag