CANopenNode/CANopenNode #61: Block Mode Transfer Issue

emanuelekms, 2017-09-25:
Dear All,

I'm working on CANopennode on dsPIC33 (PIC24_dsPIC33) custom hardware platform and I'm trying to set up SDO block mode upload. 

I'm able to perform SDO segmented transfer, but I'm having some mistakes on SDO block mode upload on server side.

I'm sending the request from a client node to a server node and I want to read a 32 byte buffer.
I've done many tests and I'm gone deep inside the library also in debug mode in order to better understand how the block mode transfer works.
In the detail I need to read a 32Byte variable ( called "bufferSpare")@ Index Ox7000 on Server Side.

At the end I'm able to perform the data transfer (SDO block mode upload) only if the variable on the Object Dictionary on Server side is defined as "VISIBLE STRING".
This is the object dictionary definition 
"{0x7000, 0x00, 0x3E, 32, (void*)&CO_OD_RAM.bufferSpare[0]},"

If I define the variable as an array of 32 elements , type unsigned 8 the SDO block mode fails (the server answer to the client as if it needs to perform a segmented mode). This seems to be connected to the lenght of data. 
If I define the variable as array the SDO manager recognizes it as leght is 1 byte (ist this the lenght of the sigle array element?) ; If I define the variable as a "VISIBLE STRING" of 32 characters the SDO manager rcognizes the legnht as 32 and the SDO block transfer is done.

Is this something I need to configure on the Object Dictionary on Server Side to make werk the SDO Block Mode transfer also for Array variables? Or is there any implementation limit that frorce me to use "VISIBLE STRING" type variables for block mode transfer?

I'll very appreciate any helps.

Best Regards 

Emanuele

martinwag, 2017-09-26:
You can only transfer one index/subindex at a time, not an entire array. If you want to do a block config download, use an additional index 'config download' with data type 'domain'.

emanuelekms, 2017-09-26:
Thank You for Your answer.
Let me better explain the what I need to do.
I need to trasnfer a large amount of data, about 1 MByte, between two nodes on CANopen bus. So my Idea is to use SDO block trasnfer mode to complete the transfer.
I noticed that max buffer size to use in an SDO transfer is 889 bytes ( #define CO_SDO_BUFFER_SIZE in CO_SDO.h file), so I realize that I need many SDO block transfer to complete 1MB data.

My first idea was to use an array of UNSIGNED8 data type with dimension = 889. 
But it seems doesn't  work and You say that I can transfer only one Index or subIndex at a time with SDO block transfer. So what is the best way to implement the transfer of large amount of data using SDO 
block transfer on CANopen?

I'm not friendly with the data type 'domain'. Could You better explain how can I use this data type for my purposes and can You give me some examples?

Thanks for Your support!

Best Regards

Emanuele

martinwag, 2017-09-26:
Domain is what you want then. Take a look in CiA301, it should be explained there. There is also some documentation in CANopenNode CO_SDO.h that explains how to use it.

JuTakata, 2017-09-26:
Hi

Emanuele : There are also example of domain on CO_Emergency I started from these examples but at this moment I only succeeded in sending 32 bytes packets.

Martin : I opened a similar discussion about Domain data type because I did not succeed to send other data size than CO_SDO_BUFFER_SIZE (32 bytes). 
That would be great if you could take a look please =)
[https://sourceforge.net/p/canopennode/discussion/387151/thread/a34fc1f3/](url)

Julien

robincornelius, 2017-09-28:
On 26 September 2017 at 09:59, JuTakata <notifications@github.com> wrote:

> Hi
>
> Emanuele : There are also example of domain on CO_Emergency I started from
> these examples but at this moment I only succeeded in sending 32 bytes
> packets.
>
> Martin : I opened a similar discussion [...]

Domain works fine

On the sending side just do a regular SDO setup, eg walk through

   CO_SDOclient_setup(CO->SDOclient,0,0,slavenode);

 CO_SDOclientDownloadInitiate(CO->SDOclient,slaveindex,slavesubindex,cantxbuf,datalen,false);

   int ret =
CO_SDOclientDownload(CO->SDOclient,time1msdiff,1000,&abortcodebuffer);

the magic is on the receiving side

set up an OD function :-
    CO_OD_configure(CO->SDO[0], 0x3005, CO_ODF_3005, NULL, 0, 0U);

in that function handle the domain :-

emanuelekms, 2017-09-28:
Dear Robin,
thank You for the reply.
I've just tried to perform an SDO block mode upload for a buffer of 889 bytes using a DOMAIN variable on the Object Dictionary.
Now I'm trying to perform the transfer of a buffer with lenght bigger tha max SDO Block lenght (889 bytes). It's not clear for me how to manage this situation using the Object Dictionary Function in order to perform multiple block transfer in SDO upload.
Have anyone tried to implement this transfer?
Any suggestion?
I'm waiting for some ideas.
Thank you for Your support.
Best Regards

Emanuele

robincornelius, 2017-09-28:
On 28 September 2017 at 13:12, Emanuele <notifications@github.com> wrote:

> Dear Robin,
> thank You for the reply.
> I've just tried to perform an SDO block mode upload for a buffer of 889
> bytes using a DOMAIN variable on the Object Dictionary.
> Now I'm trying to perform the [...]

It does not matter the SDO buffer length, this is the internal canopennode
buffer that is used to store the data as it receives it. With my previous
code i left this at 32 bytes. What happens is every time the buffer is full
OR you reach the end of the transfer the OD callback function is called and
you have to copy the data from the SDO buffer to the user buffer. You must
maintain a pointer/counter as to how many bytes you have already received
as you need to keep this pointed at the correct location in the user
buffer.   In the code i pasted i track this with a two static variables (as
you cannot run multiple SDO transfers at once on the same node this static
is fine). On the first transfer ODF_arg->firstSegment is true so it sets
the total length and resets bytesrecieved. For every other transfer it
simply copys the SDO buffer to the next position in the user buffer
until bytesrecieved ==  bytestorecieve.

One thing to watch make sure the data length in the OD is 0 for a domain
and the data pointer is NULL eg in OBJECT DICTIONARY section of CO_OD.c :-

{0x3005, 0x00, 0x0e, 0, (void*)0},

Robin

warrior9a, 2017-10-06:
Hi,

I too have been having problems with block upload domain types (transfer from / reading out of the CanOpenNode client) when the total upload length is greater than the SDO buffer length. 

It may be the way in which my application callback is implemented; I initially set dataLengthTotal, and reset firstSegment each time in my callback, as implied by the comment against the code. The problem seen was that Bit 7 in the last message of the first block wasn't being set - the behaviour set by line 1424 of SDO.c

`/* verify end of transfer */`
`if((SDO->bufferOffset == SDO->ODF_arg.dataLength) && (SDO->ODF_arg.lastSegment)){`
`    SDO->CANtxBuff->data[0] |= 0x80;`
`    SDO->lastLen = len;`
`    SDO->blksize = SDO->sequence;`
`    SDO->endOfTransfer = true;`
`}`

I have currently modified two(2) lines (first and last) of the _if_ code block to read 

`/* verify end of transfer */`
`if ( SDO->bufferOffset == SDO->ODF_arg.dataLength ){`
`    SDO->CANtxBuff->data[0] |= 0x80;`
`    SDO->lastLen = len;`
`    SDO->blksize = SDO->sequence;`
`    SDO->endOfTransfer = SDO->ODF_arg.lastSegment;`
`}`

where lastSegment is set to true in my Callback on the call containing the last segment of the whole upload.

Comments appreciated - I continue to try to understand if this is fix is required and if this is the correct fix / doesn't have undesired side effects in other modes of operation.

Comments are against https://github.com/CANopenNode/CANopenNode/tree/dee8503da0c943ced12e482f0025204a2f6dffaf

warrior9a, 2017-10-11:
Hi,

The above modification is not correct; investigating further

JuTakata, 2017-10-12:
Hi,

You modified lines on block transfer functions but domain is using segmented transfer ?

I am also trying to upload data from domain using segmented transfer. I succeeded in sending 63 bytes using 2 calls of my callback (dataLength 32bytes 1st and 31bytes 2nd) but when I try to send 64+ bytes, bytes are sent but I have a memory issue, all my OD entries are incorrect after this transfer.
I am not sure if I should use segmented or block transfer but as I understand to transfer data domain the default transfer is segmented. I do not understand the real difference between these 2 transfer at the moment but I keep learning.

Hope it helps

Julien

ololoshka2871, 2018-10-10:
I am new to CANOpen. Now I am developing a new device, or rather a bootloader for it (CiA 443). I have created an object in OD with index 0x1A50sub1, in which a blob will be written containing the main application.

Dealing with the work of the SDO blocking transfer, I came across this discussion. Maybe my approach will be useful to someone.

1. Registration of the OD-extension, which contains a callback, called when accessing the corresponding object

```cpp
static CO_SDO_abortCode_t
registerSDOExtantion(uint16_t index, const CO_OD_extension_t &extantion) {
  auto programm_od_entry = CO_OD_find(CO->SDO[0], index);
  if (programm_od_entry == 0xFFFFU) {
    return CO_SDO_AB_NOT_EXIST; // object does not exist in OD
  }
  
  CO->SDO[0]->ODExtensions[programm_od_entry] = extantion;
  return CO_SDO_AB_NONE;
}
```
2. Registration expansion. (I use the lambda function from c++11)
Reading example.
```cpp
registerSDOExtantion(
      0x1f50, // target OD index
      CO_OD_extension_t{[](CO_ODF_arg_t *ODF_arg) -> CO_SDO_abortCode_t {
        const auto total_size = 1289; // size of current DOMAIN object
        const auto remaning = total_size - ODF_arg->offset; // bytes remamning

        if (ODF_arg->firstSegment) // at first call tell library total size of this DOMAIN entry
          ODF_arg->dataLengthTotal = total_size;

        ODF_arg->dataLength = // data size to transfer at this call
            remaning > CO_SDO_BUFFER_SIZE ? CO_SDO_BUFFER_SIZE : remaning;
        // copy data to output buffer
        for (auto i = 0; i < ODF_arg->dataLength; ++i) {
          ODF_arg->data[i] = i << 1;
        }
        // set this flag, if all data writen to output buffer
        ODF_arg->lastSegment = remaning <= CO_SDO_BUFFER_SIZE;
        return CO_SDO_AB_NONE;
      }});
```

CANopenNode, 2020-12-28:
I'm not quite sure, how is with this issue now. Currently we have:
- new Object Dictionary approach, rewritten and tested SDO server in CANopenNode version 4.0
- rewritten and tested SDO client in CANopenNode version 2.0 and above
- no additional information for old Object Dictionary and old SDO server.
I'm closing this issue for now. If there will be problems with old SDO server, new issue should be opened.

closed 2020-12-28 by CANopenNode