CANopenNode/CANopenNode #62: Where do I download the Object Dictionary Editor?

xoduddk123, 2017-09-26:
Hello.

I am using the Object Dictionary Editor to create Object data for use with CANopenNode.

However, I could not find the Object Dictionary Editor in a subproject at https://github.com/CANopenNode.

I was able to find the Object Dictionary Editor at https://sourceforge.net/projects/canopennode/files/canopennode/CANopenNode-3.10/.

My idea is that the repository has been moved from sourceforge to github.
Perhaps this Object Dictionary Editor is an old version.

When I check, CO_OD.c and CO_OD.h created by Object Dictionary Editor downloaded from sourceforge are declared as **LGPL**.

[image]

However, CANopenNode is a **GPL linking exception**.
https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/example/CO_OD.c#L1-L50
https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/example/CO_OD.h#L1-L50

This confuses me.
Where can I download the Object Dictionary Editor?

I'm sorry I do not have enough English. :)

Thanks!

martinwag, 2017-10-10:
Use this one.

https://github.com/robincornelius/libedssharp

xoduddk123, 2017-10-11:
Wow this is very good.
I was inconvenient to make it with firefox.
Thanks for telling me the location of [libedssharp](https://github.com/robincornelius/libedssharp).

closed 2017-10-11 by xoduddk123

BarisYazici, 2019-06-21:
Any option for linux users?

martinwag, 2019-06-24:
Yes, I'm using the editor in Linux. It is written in C# so you can just run (and compile) it in mono.