CANopenNode/CANopenNode #63: What is the communication recovery method when the cable is reconnected?

xoduddk123, 2017-10-26:
I am very much interested in implementing CANopen with the CANopenNode project.
I am currently analyzing canopend to analyze CANopenNode, but there is one question.
This is related to operatingState when reconnecting CANopen cable.

My environment is as follows.
CANopen PLC : LS's XBL-CMEA(NodeId:127)
canopend : mcp2515 + mcp2551(NodeID:77)
[image]
[image]

I have registered canopend's CANopenSocket.eds file in XBL-CMEA.
Below is a diagram of the network situation in XG-CANopen, a related program of XBL-CEMA. (XG-CANopen is a control program of XBML-CMEA)
[image]

I have a CAN to USB ([CANtact](https://store.linklayer.com/products/cantact-v1-0?source_app=embed&referer=http%3A%2F%2Flinklayer.github.io%2Fcantact%2F)) product for debugging CANopen.
I use this USB to check the can-bus data.

Now connect the cable and run canopend to send the heartbeat signals as shown below.
[image]

Here, when I unplug the canopend cable, the following repetitive data appears.(Output data of XBL-CEMA)
[image]

If the cable is reconnected, canopend will not enter the operation state while repeating the same communication as the image below.(pre-operation state)
[https://youtu.be/IgOp8XYmlrI](https://youtu.be/IgOp8XYmlrI)
And PDO communication is not possible either.
I thought that reconnecting the cable to the canopend device would restore the operation and communicate smoothly.

In case of canopencomm 'reset communication' signal, communication is restored.

I wanted the communication to be automatically restored upon cable connection without going through canopencomm.
So I modified the source as shown below.
https://github.com/xoduddk123/CANopenSocket/commit/e01be1089338e56b6c4d790efbc90151eac0cf06
[image]

When I tested it, the communication was restored automatically when the cable was connected.
https://youtu.be/bepekqQ_NdQ

I implemented it as if I did reset comunication in canopencomm in that part for testing.
Is reset communication the only way to recover communications in this situation?
Is this the best way?

I am a beginner in canopen.
I'm sorry if it was a basic question.

I'm sorry I do not have enough English. :)
Thanks!

martinwag, 2017-11-03:
You have to separate between CAN an CANopen here.
- CAN: Recovering from Error like short is handled inside the CAN driver or inside your OS's driver. The easiest implementation is to let the hardware do it's job. Take a look at the Bosch CAN specification for details. 
- CANopen: Certain errors lead to a device leaving operational mode. This is the save state for a device. You have to set it to operational from your master again.

It's up to you how to handle those errors. As a master device, you can just go straight to operational after the CAN error has gone. After that, check all your devices for correct operation (data might have gotten lost) and set them to operational again.

xoduddk123, 2017-11-10:
I have fixed the problem as shown below.

● CANopen
Check CO-> NMT-> operatingState at regular intervals and set RESET_COMM to 0x7f (pre-op).
I used this because I found a nice function called CO_NMT_getInternalState () during my work.

● CAN
In addition, at RESET_COMM, the ip command was used to down and up the can0 interface.

Your answers have been helpful.
Thank you.

closed 2017-11-10 by xoduddk123