CANopenNode/CANopenNode #65: libedsharp CO_NO_NMT_MASTER

kenshail, 2017-11-25:
Using the excellent libedsharp EDSeditor V0.5.2 or 6 beta I have one problem: I can not get either version to set CO_NO_NMT_MASTER to 1. I am using it with the .eds file that comes with CANopenSocket. Any suggestions?

martinwag, 2017-11-27:
I think this is not supported in the GUI. Just set it manually inside the header files.

robincornelius, 2017-11-27:
It does but it requires the appropriate master bits to be set in the OD

It requires Object 0x1f80 to exist and the value of object 0x1f80 must have
bit 0 set (eg have a value of 0x01)

See
https://github.com/robincornelius/libedssharp/blob/5c8f76f75f49c63db1757c9a359dc50c61b9184b/libEDSsharp/CanOpenNodeExporter.cs#L399

Regards

Robin

kenshail, 2017-11-27:
Thank you Robin. That works fine.
 And thank you for the excellent editor.

closed 2017-11-27 by kenshail