CANopenNode/CANopenNode #69: GPL2 license

nilesh441, 2017-12-14:
I apologize if I posted this question at wrong place. I tried to find your email address but not succeeded. So posted a question here. 

I want to use your code in binary form in my commercial product. Can I use it? I have read the terms of GPL2 but it is confusing so I decided to ask you directly.

w1ne, 2018-06-28:
Please see discussion [here](https://sourceforge.net/p/canopennode/discussion/387151/thread/b7910450/?limit=25#13fd)
You can use CANopenNode in commercial products as separate library. Just follow licence rules and pull back any changes to the stack. Your application code and library code can have separate licenses according Classpath Exception.

djleach-belcan, 2018-08-22:
I'm not seeing the quoted classpath exception in the [LICENSE](https://github.com/CANopenNode/CANopenNode/blob/master/LICENSE)file. Am I missing something?

henrikbrixandersen, 2019-09-20:
It would be nice to clarify if the CO_OD.{h,c} files are considered part of the stack or part the application license wise.

CANopenNode, 2019-09-21:
Commented here:
https://github.com/CANopenNode/CANopenNode/issues/105#issuecomment-533803323

closed 2019-10-08 by martinwag