CANopenNode/CANopenNode #71: Variable-Length Strings

iurly, 2018-01-06:
I'm trying to implement a few variable-length strings in the Object Dictionary (read-only, so SDO upload).
What's the recommended way of going about it?

@martinwag: what I'm trying to do is read (among other things) the software version just like you posted [here](https://sourceforge.net/p/canopennode/discussion/387151/thread/258973ec/), would you care to share how you did that? Did you implement some custom OD function or do you just patch CO_OD.c before compiling?

Thank you!

martinwag, 2018-01-15:
I've reserved the space for the longest possible string.

so \0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0\0
in OD editor for Git
V65535.65535-999-12345678-dirty

The version string is then inserted on CANopen startup by firmware.

iurly, 2018-01-15:
I see, thanks. But how do you handle the variable-length? It turns out in that scenario CANopenNode would return a string which is just as long as the longest possible string, padded with \0's at the end.
So in the end I settled for some custom function.

martinwag, 2018-01-15:
Yes, that's the case. I don't relly care for that as the versions are the only strings in my application.

closed 2020-01-28 by CANopenNode