CANopenNode/CANopenNode #72: Multiple(xed) PDOs

iurly, 2018-01-22:
Hi,

I need for a single node to autonomously transmit several process data (i.e. more than 4 PDOs * 8 bytes/PDO).
In my understanding there's a limit for each node to only send 4 different PDOs, and CANOpen already defines multiplexed PDOs to address such a case (at the cost of extra bandwidth of course).
Is there any plan or idea to implement such a feature in CANopenNode?

Would there be any alternative (e.g. by using custom COB-IDs?) I wouldn't need to deploy a huge number of nodes so I might as well restrict the usage of NodeIDs to e.g. 8,16,24 etc... so that the node with ID 16 can use up all the COB-IDs that would normally be used by 8 different nodes (with ID between 16 and 23).
Any suggestion welcome...
Thank you!

robincornelius, 2018-01-22:
On 22 January 2018 at 09:50, iurly <notifications@github.com> wrote:

> Hi,
>
> I need for a single node to autonomously transmit several process data
> (i.e. more than 4 PDOs * 8 bytes/PDO).
> In my understanding there's a limit for each node to only send 4 different
> PDOs, [...]

There is no actual limit to the number of TX PDOs a device can use (or the
limit is in fact very large as you can use any COB-ID in the PDO space),  I
think what you are thinking of is the 4 default TX PDOs 0x180+$NODEID -
0x480+$NODEID

> Would there be any alternative (e.g. by using custom COB-IDs?) I wouldn't
> need to deploy a huge number of nodes so I might as well restrict the usage
> of NodeIDs to e.g. 8,16,24 etc... so that the [...]

Exactly this, use any COB-IDs (in the TX PDO range) you like, you just need
to ensure you have no bus conflicts where two devices both use the same TX
PDO. If you do have a TX PDO conflict you will not notice until both
devices transmit a slightly different packet at exactly the same time and
then both have a NAK fight.

Robin

mike48436, 2019-11-22:
Hi Iurly
did you try to send with different COB_ID from the same slave? how did you do this? is there a release of it? At the moment I'm trying the same with the same code from "https://github.com/CANopenNode/CANopenNode" but I'm still struggling.

Thank you for your help.

CANopenNode, 2020-01-27:
I hope it is solved now. CANopen device has 4 TPDOs by default, but this can me extended to much more. Just edit the Object dictionary and set custom COB-IDs.

The other alternative use of SDO. SDO with domain data type can transfer really large amount of 
data.

Multiplex PDO (MPDO) also exists in CANopen, but it is currently not implemented.

closed 2020-01-28 by CANopenNode

bullitmaster, 2023-06-15:
Hi, all!
We use R PDO cob-id (0x200, 0x300, 0x400 and 0x500) for transmit cob-id)

ks1930, 2025-07-15:
@bullitmaster as canopennodedoes not support mpdo did you guys then somehow restrict the node from sending the id along with cob id 0x200 , 0x300 or 0x 400 fror MPDOs? , how did you implement it ?