CANopenNode/CANopenNode #73: Avoiding multibyte object split

kingd123, 2018-02-08:
Hi folks,

CO_PDO.c lock/unlock as below are needed, no ?

int16_t CO_TPDOsend(CO_TPDO_t *TPDO){
:
    /* Copy data from Object dictionary. */
    CO_LOCK_OD();
    /* Prevent mid-use multibyte object dictionary item alter by interrupting
    /* item write */
    for(; i>0; i--) {
        *(pPDOdataByte++) = **(ppODdataByte++);
    }
    CO_UNLOCK_OD();  /* Allow object dictionary item alter */
:
}

This to prevent for example, interrupt when only _one_ byte of a uint16 object has been copied,
_then_ an interrupting user-code function _updates_ the uint16 in the OD, _then_ the above code
resumes and copies the second byte. Resulting in, the sent uint16 containing comprising an
old and new byte, which do not necessarily form a coherent whole for the usage in question.
At least, if you have an interrupting routine that writes to the OD, as I have.

Of course the same applies to any interrupted multibyte object copy. Such as a float, or a string
(max 8 bytes): If part old bytes and part new bytes, are copied, the resultant sent whole might not
necessarily be coherent. Or for float, even valid.

In CO_driver.h, lock/unlock macros defined for example thus:

#define CO_LOCK_OD()   __disable_irq( );  /**< Lock critical section when accessing Object Dictionary */
#define CO_UNLOCK_OD() __enable_irq( );   /**< Unlock critical section when accessing Object Dictionary */

Best regards,

David

martinwag, 2018-02-16:
Yes, they are needed, as this is basically a multithreaded program. If you go single threaded (meaning no interrupts), you don't need any locks.

closed 2019-10-08 by martinwag

MartySchrader, 2021-09-21:
While I would guess it's okay to label anything with separate interrupt handlers as "multithreaded," the issue of split precision can also be addressed with semaphores or mutexes. For instance, if your background operation raised a semaphore every time it updated a value then the front end would always know if it had the latest value or not. As such:

Do
  Done = false
  Semaphore = false
  Copy value
  If (Semaphore is still false) then
    Done = true
While (Done is not true)

This is how the front end knows it has the latest, non-buggered-up value.