CANopenNode/CANopenNode #74: Overrun ?

kingd123, 2018-02-12:
Hi folks,

I'm trying to understand the usage intent, of the following code, within drvTemplate > CO_driver.c.

In very general terms, the design intent would seems to be send the message if the CAN CPU hardware is free (with the 1 in 1 && to be replaced by a check that it is, as the supplied CPU drivers have done).

But if not free, set the buffer->bufferFull flag, indicating that the buffer in question has a pending message to be sent, in due course by interrupt. For example, TPDO->CANtxBuff>bufferFull.

And that it how I find the stack to be operating - I have it all up and running, on an Infineon XMC4 CPU.

Just one problem, though: When the function below is again called, if the buffer->bufferFull flag is set, 
a CO_EMC_CAN_OVERRUN EMCY is generated. And indeed I see this EMCY, with a CANbus analyser. But it seems to me, this EMCY doesn't need to be generated. Because there hasn't really be an overrun and loss of message. Just, a message is waiting to be sent by interrupt.

But perhaps, I've got something wrong with my understanding, Can anyone shed some light ?

Infact, I this CO_EMC_CAN_OVERRUN is constantly occurring. But of course, I only get the one EMCY.

A consequence, is that the 1003 Pre-defined error field, changes from 0 to 0x10, signifying comms errors. As a result, NMT blocks transition of my device to Operational mode, from Pre-Op or Stopped.

It goes OK to Operational at boot or on NMT comms reset, though.

Thanks,

David King

```
CO_ReturnError_t CO_CANsend(CO_CANmodule_t *CANmodule, CO_CANtx_t *buffer){
    CO_ReturnError_t err = CO_ERROR_NO;

    /* Verify overflow */
    if(buffer->bufferFull){
        if(!CANmodule->firstCANtxMessage){
            /* don't set error, if bootup message is still on buffers */
            CO_errorReport((CO_EM_t*)CANmodule->em, CO_EM_CAN_TX_OVERFLOW, CO_EMC_CAN_OVERRUN, buffer->ident);
        }
        err = CO_ERROR_TX_OVERFLOW;
    }

    CO_LOCK_CAN_SEND();
    /* if CAN TX buffer is free, copy message to it */
    if(1 && CANmodule->CANtxCount == 0){
        CANmodule->bufferInhibitFlag = buffer->syncFlag;
        /* copy message and txRequest */
    }
    /* if no buffer is free, message will be sent by interrupt */
    else{
        buffer->bufferFull = true;
        CANmodule->CANtxCount++;
    }
    CO_UNLOCK_CAN_SEND();

    return err;
}
```

iurly, 2018-02-13:
Hi David,

I had the exact same problem and was about to ask the same question.
My understanding is that the driver somehow implements some sort of 1-element queue per CANopen "entity" (i.e. SDO, NMT, each TPDO etc..).
So every time each entity needs to send a CAN frame (which it will kindly write to its own personal CO_CANtx_t buffer), three things can happen:
1) the hardware is ready and has a free TX slot available (which in the template driver is denoted by CANtxCount == 0, meaning there are no other queued buffers to send). So it will just request transmission and wait for the interrupt to signal that TX was successful.
2) the hardware is currently transmitting and/or there are other messages queued for sending (CANtxCount !=0), but **NOT** from the current entity (bufferFull == false). In this case, you're allowed to queue up **ONE** message for this entity, by just setting bufferFull = true.
3) same as 2), but there's also one message from this same entity (same CO_CANtx_t, bufferFull == true). In this case you trigger an overrun.

In my understanding, the rationale behind this is as follows: if you want to send periodic data (e.g. TPDOs), you want to do it at a rate that's sustainable for the whole bus. This means, by the time you try sending the same TPDO *again* (a new one, with new data), the previous one must have made it to the bus, otherwise your data rate is just not sustainable.
This may or may not be what you want to pursue, but it could make sense in a context where timing is so crucial (i.e. industrial control applications?).
In my case, which is just a logging application for sensors, it is way more important that no data ever gets lost, while a small delay doesn't really cause any problem.

So I actually came up with the following (simple TX queue), please have a look and feel free to comment and/or take it:
https://github.com/iurly/CANopenNode/commit/6349ea6d1dfdb01031bac829d5fd1698df26a242

The codebase itself is quite ugly but the commit itself should be usable/intelligible enough.
Feedback welcome.

martinwag, 2018-02-14:
This is an implementation that means no data can be lost without notice. Depending on your use case, your requirements could be more relaxed. E.g. if you can accept that only the _last_ TPDO value will be sent:
- make CO_CANtxBufferInit() aware of TPDOs
- when you try to write a TPDOs that is marked "bufferFull", just replace the current content.

Of course, there are more solutions to this, like replacing the one-element queue with a proper queue (see iurlys post for that). In my implementation, i completely ditched "bufferFull", and rely instead on the OSses queue for that (like linux implementation).

regards,
Martin

martinwag, 2019-10-08:
I think question is answered.

closed 2019-10-08 by martinwag