CANopenNode/CANopenNode #81: CO_init should take CANbaseAddress as void *

Amomum, 2018-04-19:
Now CO_init looks like this:

```
CO_ReturnError_t CO_init(
        int32_t                 CANbaseAddress,
        uint8_t                 nodeId,
        uint16_t                bitRate);
```
However, CANbaseAddress usually is an address of some kind of structure, for example, in stm32 it's a pointer to `CAN_TypeDef`.
So casting it to and from `int32_t `is:

- ugly, especially in C++
- may be undefined behaviour if size of pointer is bigger than `int32_t`

Since CANbaseAddress is used only for forwarding to CO_driver and not for it's numerical value, correct way of passing it will be as `void *`.

odesenfans, 2018-04-23:
I have a branch on my fork with a basic version of the fix (https://github.com/odesenfans/CANopenNode/tree/can-base-address-void-ptr), will try to push it when the maintainers are active. I just fixed the socketCAN port for now.

roman-khazanskii, 2018-04-23:
@odesenfans that's great!

wilkinsw, 2019-11-05:
I expect this was done because some embedded platforms provide an actual address rather than a pointer.  For example on my Kinetis MCU I have this:
```c
/* CAN - Peripheral instance base addresses */
/** Peripheral CAN0 base address */
#define CAN0_BASE                                (0x40024000u)
```

Which I can pass into CO_init directly and then cast properly in CO_driver.c
```c
 CAN_Type *base = (CAN_Type *)CANbaseAddress;
```

There is also the socketcan implementation.  In CANopenSocket the output of `if_nametoindex` (an unsigned int) is passed as the CANbaseAddress.

That said, passing a void* seems like it would be the most flexible approach.  @odesenfans do you want to update your branch and open a pull request?  Development has resumed.

odesenfans, 2019-11-06:
Hi @wilkinsw , glad to see that the project is moving again! I'll rebase my branch over the weekend and update it accordingly.

odesenfans, 2019-11-10:
There you go, PR open! https://github.com/CANopenNode/CANopenNode/pull/116

CANopenNode, 2020-01-27:
Solved long time ago ....

closed 2020-01-27 by CANopenNode