CANopenNode/CANopenNode #82: Getting Started

Yatekii, 2018-06-13:
Hi Folks!

I am just about to getting started using this lib. I had a good look at the lib and have a few questions before I start.

1. If I understand correctly, this lib provides the CANopen machinery and provides hooks where I can add in my own drivers. How to do this I can find in the `drvTemplate`. Correct?

2. I want to use the lib on an STM32F072. There is two drivers for STM32's. Can I simply use the generic one or am I better off writing my own?

3. To autogenerate the Object Database, a "Web Application" is needed. I figured that I can only use it in IE (ohwow.). But even in IE I get some errors and are unable to generate a project. How do I create a new project?

4. The project is completely unmaintained. But there are a lot of pull requests and apparently interested people. Why not just create a fork of the project and have it maintained by some active people and not just one person? I would really love to use this lib and also contribute but with it largely unmaintained it's kind of bad ...

Thanks a lot for any answers & best regards,
Yatekii

Cking616, 2018-06-13:
1,2. You need to write a driver yourself based on your development pin usage,Imitate the drvTemplate file or stm32 provided write, and added to the CAN interrupt function.
3.You can use https://github.com/robincornelius/libedssharp to autogenerate the Object Database.

Yatekii, 2018-06-14:
Thanks a ton! I will try with libedssharp :)

martinwag, 2019-10-08:
I think question is answered.

closed 2019-10-08 by martinwag