CANopenNode/CANopenNode #86: Dereference rcvMsg

JuPrgn, 2018-09-04:
Hi,

I am using dsPIC33 CO_driver and using code analysis I found the following error message : 

"Null pointer dereference: rcvMsg"

I can't figure out how is it working : 

`CO_CANrxMsg_t *rcvMsg;      /* pointer to received message in CAN module */
...
        rcvMsg = 0; /* get message from module here */
        rcvMsgIdent = rcvMsg->ident;`

The stack is works fine on my project but on this code rcvMsg is set to a NULL address and is used to fill rcvMsgIdent. Can you please help me to understand this part ?

Here is the full code : 

`void CO_CANinterrupt(CO_CANmodule_t *CANmodule){

    /* receive interrupt */
    if(1){
        CO_CANrxMsg_t *rcvMsg;      /* pointer to received message in CAN module */
        uint16_t index;             /* index of received message */
        uint32_t rcvMsgIdent;       /* identifier of the received message */
        CO_CANrx_t *buffer = NULL;  /* receive message buffer from CO_CANmodule_t object. */
        bool_t msgMatched = false;

        rcvMsg = 0; /* get message from module here */
        rcvMsgIdent = rcvMsg->ident;
        if(CANmodule->useCANrxFilters){
            /* CAN module filters are used. Message with known 11-bit identifier has */
            /* been received */
            index = 0;  /* get index of the received message here. Or something similar */
            if(index < CANmodule->rxSize){
                buffer = &CANmodule->rxArray[index];
                /* verify also RTR */
                if(((rcvMsgIdent ^ buffer->ident) & buffer->mask) == 0U){
                    msgMatched = true;
                }
            }
        }
        else{
            /* CAN module filters are not used, message with any standard 11-bit identifier */
            /* has been received. Search rxArray form CANmodule for the same CAN-ID. */
            buffer = &CANmodule->rxArray[0];
            for(index = CANmodule->rxSize; index > 0U; index--){
                if(((rcvMsgIdent ^ buffer->ident) & buffer->mask) == 0U){
                    msgMatched = true;
                    break;
                }
                buffer++;
            }
        }

        /* Call specific function, which will process the message */
        if(msgMatched && (buffer != NULL) && (buffer->pFunct != NULL)){
            buffer->pFunct(buffer->object, rcvMsg);
        }

        /* Clear interrupt flag */
    }`

Regards

martinwag, 2018-09-05:
You've quoted the driver template. Exclude the files you don't use from your analysis.

JuPrgn, 2018-09-05:
Hi Martin,
Thank you, I was stuck on analysis and didn't realized this was the template ! =)

closed 2018-09-05 by JuPrgn