CANopenNode/CANopenNode #89: CANopen and stm32 with HAL

Barto22, 2018-09-07:
Hi,
I'm new to CANopen node stack, I want to make it work with stm32 but I saw that the CO_driver is based on the standard library and I want to use HAL. Please help me guys what should I change and where should I start or if someone have working code for some stm32 with HAL could you send me it? At the beginng I want to make some kind of example like it is for PIC.
Best regards

TheLamy, 2019-04-17:
Does anybody have a solution for this now? I'm also trying to write the driver on my own, but I'm struggling with it.

StefanoPietrosanti, 2019-04-17:
I was planning on writing the driver later this year, I need to find the time for it. It is definitely something that will be useful for the new STM32 generation.

ad7718, 2019-08-12:
https://github.com/w1ne/CANOpenNode-CubeMX-HAL

martinwag, 2019-10-08:
I think question is answered.

closed 2019-10-08 by martinwag

Mutharagan, 2020-04-15:
> I ported this https://github.com/w1ne/CANOpenNode-CubeMX-HAL to stm32f042, works fine. I only changed can instance and recalculated clocks regs for can.

Hi, 
I have included CANopen files with my project, some doubts,  have you mentioned CO_configurationMode() . What are the things need to be put in that function definition.

martinwag, 2020-04-20:
Please open an new issue. No one will see messages added to a closed issue.