CANopenNode/CANopenNode #90: Compilation Error

e135193, 2018-09-28:
I am getting an error for the below code line.

`buffer->pFunct = pFunct;`

`Error[Pe513]: a value of type "void (*)(void *, CO_CANrxMsg_t )" cannot be assigned to an entity of type "void ()(void *, CanRxMsgTypeDef *)"`

Do you have the full CubeMx project that I can compile without a problem ?

e135193, 2018-10-03:
Where should I put my CanTx() and CanRx() functions in the code ?

martinwag, 2018-10-04:
Read the comments in the driver template. You need at least basic understanding as just putting rx/tx will not work.

e135193, 2018-10-07:
I am able to port canopennode to new HAl/Cube version. However I have a problem. The node start sending heartbeat at first then stop sending any data. below is the can output.

What could be the problem ?

Sample Output #1
701 00 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
081 00 00 00 01 41 00 60 00
081 20 81 00 07 33 00 87 00
081 20 81 00 07 33 00 87 00
081 20 81 00 07 33 00 87 00
081 20 81 00 07 33 00 87 00
081 20 81 00 07 33 00 87 00
081 20 81 00 07 33 00 87 00
081 20 81 00 07 33 00 87 00
081 20 81 00 07 33 00 87 00
081 20 81 00 07 33 00 87 00
081 20 81 00 07 33 00 87 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00

Sample Output #2
701 00 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
701 05 00 00 00 00 00 00 00
081 20 81 00 07 33 00 84 02
081 20 81 00 07 33 00 84 02
081 20 81 00 07 33 00 84 02
081 20 81 00 07 33 00 84 02
081 20 81 00 07 33 00 84 02
081 20 81 00 07 33 00 84 02
081 20 81 00 07 33 00 84 02
081 20 81 00 07 33 00 84 02
081 20 81 00 07 33 00 84 02
081 20 81 00 07 33 00 84 02
081 20 81 00 07 33 00 84 02
081 20 81 00 07 33 00 84 02
081 20 81 00 07 33 00 84 02
081 20 81 00 07 33 00 84 02
081 20 81 00 07 33 00 84 02
081 20 81 00 07 33 00 84 02

closed 2018-10-12 by e135193