CANopenNode/CANopenNode #91: CO_EMC_BUS_OFF_RECOVERED, Emergency

e135193, 2018-10-09:
Hello,
I successfully ported the CanOpenNode to an STM32 device. However, there is a problem. I get the emergency message which is CO_EMC_BUS_OFF_RECOVERED. 

What could be the reason for this?

Node ID is 2.

Output from Node:
082 40 81 10 12 47 00 fd 00
082 40 81 10 12 47 00 fd 00
082 40 81 10 12 47 00 fd 00
082 40 81 10 12 47 00 fd 00
082 40 81 10 12 47 00 fd 00
082 40 81 10 12 47 00 fd 00
082 40 81 10 12 47 00 fd 00
082 40 81 10 12 47 00 fd 00
082 40 81 10 12 47 00 fd 00
082 40 81 10 12 47 00 fd 00
082 40 81 10 12 47 00 fd 00
082 40 81 10 12 47 00 fd 00

gotocoffee1, 2018-10-11:
[This ](https://github.com/CANopenNode/CANopenNode/blob/master/stack/STM32/CO_driver.c#L398)is the only place where is occures. But I don't use that MCU so I can't help.

martinwag, 2018-10-22:
Have a look on the Bosch CAN specification. This explains what leads to bus off.

You only see the CO_EMC_BUS_OFF_RECOVERED because the CO_EMC_BUS_OFF_ACTIVE message can't be sent in bus off state...

closed 2018-10-23 by e135193