CANopenNode/CANopenNode #97: CANopenNode on MBED os 5

geobruce, 2019-04-05:
Hi

I'm working with MBED os 5 on a STM32F091RC (NUCLEO-F091RC) and I'm searching for a CANopenNode implementation which is compatible with MBED os 5.

Or is it possible to just use this configuration: https://github.com/CANopenNode/CANopenNode/blob/master/stack/STM32/CO_driver.c ? 

I'm unsure because on the homepage in the table: Microcontroller support there is no status known for STM32.

All tips and tricks are welcome.

Thank you in advance.

WillyKaze, 2019-04-08:
Hi, 

If you need any pointers to get you started, you can have a look at the LPC1768 implementation which is compatible with Mbed 5.

https://github.com/CANopenNode/CANopenNode/blob/master/stack/LPC1768/CO_driver.cpp

geobruce, 2019-04-08:
> Hi, If you need any pointers to get you started, you can have a look at the LPC1768 implementation which is compatible with Mbed 5. https://github.com/CANopenNode/CANopenNode/blob/master/stack/LPC1768/CO_driver.cpp 

Thank you I'll have a look at it. Is there somewhere else some documentation on how to use the library?

geoffrey-vl, 2019-10-07:
Issue is covered here: https://github.com/Alphatronics/CANopenNode/tree/mbedSTM32

please close

martinwag, 2019-10-08:
please add your driver to the wiki https://github.com/CANopenNode/CANopenNode/wiki when finished, so others can find it.

closed 2019-10-08 by martinwag