CANopenNode/CANopenNode #98: RTR for  TPDO??

mjphillips1981, 2019-04-18:
Thank you so much for this CANopen library, it has made my life much easier implementing it into my design.  I pretty much have everything running, but when I try to RTR my TPDO, I get no response.  I believe it has something to do with the filters/masks??  Not real sure.  I have been digging for clues for the past few days on this.  Any help would be appreciated.

Thanks,
Mike

robincornelius, 2019-04-19:
On Thu, 18 Apr 2019 at 19:06, mjphillips1981 <notifications@github.com>
wrote:

> Thank you so much for this CANopen library, it has made my life much
> easier implementing it into my design. I pretty much have everything
> running, but when I try to RTR my TPDO, I get no response. I [...]
Did you set the TXPDO communication parameters.type to either 252 or 253
which is required for the RTR function to work. 252 requires a SYNC packet
and an RTR

Robin

mjphillips1981, 2019-04-19:
Robin,  

Thank you for your response.  Unfortunately, it does not work either.  This stack doesn't seem to support RTR messages.  I am trying to design a CANopen angle sensor.  After testing a competitors unit, all I have to do is send and RTR to the TPDO and it responds with the TPDO message.  The transmission type on the device is set to 254.  Maybe this has to be built custom?

Thanks,
Mike

follower, 2019-04-22:
As far as I can tell, RTR functionality is explicitly & intentionally unsupported:

 * *"[Responding to the queries from the CanOpen master is not implemented. This technique uses RTR CAN frames from master. RTR (remote transmission request) is special CAN frame. CiA recommends not to use RTR.](https://sourceforge.net/p/canopennode/discussion/387151/thread/0658e382/#ea68)"*

 * See also: "[How to read TPDO with remote request](https://sourceforge.net/p/canopennode/discussion/387151/thread/f33cd09a/)", "[Use of RTR bit by CANOpenNode ](https://sourceforge.net/p/canopennode/discussion/387151/thread/2a4a5b04/)", "[NodeGuarding](https://sourceforge.net/p/canopennode/discussion/387151/thread/4a8329ed/)".

This code prevents transmission type of 252 or 253 ([the values for RTR according to this reference](https://www.canopensolutions.com/english/about_canopen/pdo.shtml)) being accepted:

https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/stack/CO_PDO.c#L559-L561

martinwag, 2019-10-08:
see #99

closed 2019-10-08 by martinwag