CANopenNode/CANopenNode #99: Clarify that RTR is unsupported by CANopenNode

follower, 2019-04-22:
It would be helpful to clarify here that transmission types 252 & 253 are valid according to the specification but explicitly not implemented/supported by CANopenNode:

https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/stack/CO_PDO.c#L559-L561

For additional background see: https://github.com/CANopenNode/CANopenNode/issues/98#issuecomment-485402918

follower, 2019-04-22:
Also here: https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/stack/CO_PDO.h#L140

mjphillips1981, 2019-04-22:
Thank you.  I found those lines of code as well.  I am thinking that the competitor unit that I have is using some custom firmware to handle their RTR response for the TPDO.  Because the device does not need to be set to 252 or 253 to use RTR.  I am just going to omit this in my device.  The same results can be acquired using the SYNC method.

closed 2020-01-10 by CANopenNode