CANopenNode/CANopenNode #100: TPDO Calls Extension use of ODF_arg doesn't match SDO OD function ODF_arg behaviour

follower, 2019-04-25:
### Context

I first encountered this issue because the current implementation of the `TPDO_CALLS_EXTENSION` doesn't handle the Record type correctly (and attempted modification of `.data` within an Object Dictionary Function causes a crash as a result). 

Further investigation lead to the realization that there seem to be multiple potential issues with the current approach. (I suspect `RPDO_CALLS_EXTENSION` may be affected also.)

### Initial cause

When I first tested the TPDO Object Function feature extension with modification of `readInput8Bit` it worked but later with `testVar` a crash occurred. My understanding is this is because `TPDO_CALLS_EXTENSION` uses only the direct value of the data pointer:

https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/stack/CO_PDO.c#L885

...rather than having the pointer value interpreted based on the data type--as implemented in the `CO_OD_getDataPointer()` function:

 https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/stack/CO_SDO.c#L524-L550

### Workaround

For my purposes I worked around the issue by setting `ODF_arg.data` via a call to `CO_OD_getDataPointer()`:

```patch
-            ODF_arg.data = pSDO->OD[entryNo].pData;
+            ODF_arg.data = CO_OD_getDataPointer(pSDO, entryNo, subIndex);
```

But this isn't actually consistent with the documented behaviour either--the documentation for the `ODF_arg` struct mentions`ODF_arg.ODdataStorage` is supposed to point to the OD (and not be modified) and `ODF_arg.data` is supposed to point to the "SDO data buffer" and be modifiable:

https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/stack/CO_SDO.h#L548-L557

### Improved solution possibility

I think a more correct approach could be possible via use of `CO_SDO_initTransfer()` and/or `CO_SDO_readOD` to perform the initial population of `ODF_arg` as they seem to handle the population correctly.

In addition, my understanding is that the OD content is supposed to be copied to the transmit buffer first and *then* the Object Dictionary Function called.

### Further relevant references

* `CO_ODF_arg_t` struct: https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/stack/CO_SDO.h#L538-L586

* `CO_OD_entry_t` struct documentation for `pData` member: https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/stack/CO_SDO.h#L512-L518

* Code for handling Record type in `CO_OD_getDataPointer()`: https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/stack/CO_SDO.c#L547-L548

* Full `CO_TPDOsend()` function: https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/stack/CO_PDO.c#L856-L903

* `CO_SDO_initTransfer` function: https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/stack/CO_SDO.c#L568-L614

* `CO_SDO_readOD` function: https://github.com/CANopenNode/CANopenNode/blob/533801574055c5dacf60fb6b18930c288336dce1/stack/CO_SDO.c#L618-L679

* See also: `CO_SDO_writeOD` function.

* Original `TPDO_CALLS_EXTENSION` patch/discussion thread: https://sourceforge.net/p/canopennode/discussion/387151/thread/904bbabe/

* Original `TPDO_CALLS_EXTENSION` commit/patch: <https://sourceforge.net/p/canopennode/code_complete/ci/ef87f6eca4c7b6c814ca963eedb278715f8320d5/>

[This isn't a complete bug report but I wanted to at least document the issue for others to find and link to potential solutions for myself or someone else to follow up later.]

gotocoffee1, 2019-05-02:
I can confirm this the error occurs also in RPDOs. I did the same to fix the problem and it worked.

martinwag, 2019-05-06:
I can confirm this, too. I also don't have a good solution for this...

closed 2020-01-28 by CANopenNode