CANopenNode/CANopenNode #101: Low power application with dsPIC33E

mjphillips1981, 2019-04-30:
Hello all,
I was just wondering if anyone has ever implemented this library in a low power design.  I am using the TI sn65hvd230 transceiver which can be put into low power mode by pulling its Rs pin high.  This drops the current draw from the chip dramatically.  I would like to use this feature but I am not sure when to turn on and off the transceiver.  I have tried several places throughout the code, but I get all kind of bus errors and then a bus off situation.  Any help or tips would be appreciated!

Thanks,
Mike

martinwag, 2019-05-06:
I don't know this tranceiver. Generally, you need to be RX ready at any time and TX ready when a CAN message needs to be sent. So check that using low power mode doesn't disable RX and switch on transceiver TX before sending. Check for minimum time before disable power-save and first message.

martinwag, 2019-10-08:
I think question is answered.

closed 2019-10-08 by martinwag