CANopenNode/CANopenNode #102: Bad NMT messages

heiko-jakob, 2019-07-23:
The NMT Messages are buggy. Length of the package is always 0

can0  701   [0] 

should be something like

 can0  701   [1]  00
  can0  701   [1]  05
  can0  701   [1]  7F

Bugfix is to set the correct DLC in stack/CO_NMT_Heartbeat.c

$ git diff
diff --git a/stack/CO_NMT_Heartbeat.c b/stack/CO_NMT_Heartbeat.c
index cc4b1d5..4562b40 100644
--- a/stack/CO_NMT_Heartbeat.c
+++ b/stack/CO_NMT_Heartbeat.c
@@ -229,6 +229,7 @@ CO_NMT_reset_cmd_t CO_NMT_process(
         NMT->HBproducerTimer = 0;
         NMT->HB_TXbuff->data[0] = NMT->operatingState;
+        NMT->HB_TXbuff->DLC = 1;
         CO_CANsend(NMT->HB_CANdev, NMT->HB_TXbuff);
         if(NMT->operatingState == CO_NMT_INITIALIZING){

heiko-jakob, 2019-07-23:
` git diff
diff --git a/stack/CO_NMT_Heartbeat.c b/stack/CO_NMT_Heartbeat.c
index cc4b1d5..4562b40 100644
--- a/stack/CO_NMT_Heartbeat.c
+++ b/stack/CO_NMT_Heartbeat.c
@@ -229,6 +229,7 @@ CO_NMT_reset_cmd_t CO_NMT_process(
         NMT->HBproducerTimer = 0;
 
         NMT->HB_TXbuff->data[0] = NMT->operatingState;
+        NMT->HB_TXbuff->DLC = 1;
         CO_CANsend(NMT->HB_CANdev, NMT->HB_TXbuff);
 
         if(NMT->operatingState == CO_NMT_INITIALIZING){
`

kesboh, 2019-08-02:
I think you have an error in your driver. In CO_NMT_heartbeat.c we have this, under the `CO_NMT_init()` function:

`
    NMT->HB_TXbuff = CO_CANtxBufferInit(
            HB_CANdev,          /* CAN device */
            HB_txIdx,           /* index of specific buffer inside CAN module */
            CANidTxHB,          /* CAN identifier */
            0,                  /* rtr */
            1,                  /* number of data bytes */
            0);                 /* synchronous message flag bit */
`
As you can see the DLC is set to 1, and as it is never again modified it should stay constant.  The `CO_CANtxBufferInit()` function is in your CO_driver.c file, which is why I'm guessing that's where you have introduced the bug.

martinwag, 2019-10-08:
kesboh is correct. I think question is answered.

closed 2019-10-08 by martinwag