CANopenNode/CANopenNode #103: CANopen for Tiva C TM4C123GXL

hkn1304, 2019-08-28:
Hello gents,

May i ask what needs to be changed to make it compatible with TM4C123GXL? Is it just the can configuration?

robincornelius, 2019-08-29:
Essentialy you need to implement your own CO_Driver.c as this is the
interface between the stack and the underlying hardware. There are about 8
functions that need implementing to configure the
hardware/send/recieve/setting up filters etc.

You also need to implement a 1ms timer and call a couple of functions from
there and also handle the can interupt and again call the stack from there.
The PIC32/main_PIC32.c is a pretty good example of how to init the stack
and where to put the timer/interupt handler and most of the code in that
file is generic, only the timer/interrupt bits are processor specific.

If you use eeprom storage you also need to implement your own eeprom.c
file, again find one of the exisiting examples there are two bits to it,
the standard bits that need supporting for handing objects 1010 and 1011
and the rest of the file is specific to your hardware/storage solution.

Robin

martinwag, 2019-10-08:
I think question is answered. please add to https://github.com/CANopenNode/CANopenNode/wiki when your driver is finished.

closed 2019-10-08 by martinwag