CANopenNode/CANopenNode #104: Minimum data length of SDO uploads

DaanConn, 2019-09-11:
If one needs to run an SDO upload (read an OD entry from another node), you may do so by calling function CO_SDOclientUploadInitiate(). Initially, this function checks the input parameters:

```
if(SDO_C == NULL || dataRx == 0 || dataRxSize < 4) {
    return CO_SDOcli_wrongArguments;
}

```
However, I don't understand why the dataRxSize must be greater than 4, an OD entry may contain 1, 2 or 4 bytes. Why must it be larger than 4? Anyone a suggestion?

Daan

robincornelius, 2019-09-11:
On Wed, 11 Sep 2019 at 08:20, DaanConn <notifications@github.com> wrote:

> If one needs to run an SDO upload (read an OD entry from another node),
> you may do so by calling function CO_SDOclientUploadInitiate(). Initially,
> this function checks the input parameters:
>
> if(SDO_C == NULL || dataRx == 0 || dataRxSize < 4) {
>     return CO_SDOcli_wrongArguments;
> }
>
>
> However, I don't understand why the dataRxSize must be greater than 4, an
> OD entry may contain 1, 2 or 4 bytes. Why must it be larger than 4? Anyone
> a suggestion?
>
>
> Because the SDO expedited transfer buffer is always 4 bytes, regardless if
you use them all or not. If the SDO changes to an ABORT message all bytes
are used to fill in the error. If CO_SDOclientUpload() is < 0 then read
your buffer 4 bytes to see the error code.

Robin

DaanConn, 2019-09-11:
Thank you Robin, that makes sense. Just one more thing, the data size should be bigger than 4, so 5 bytes or more, however, in case of an abort, the error code is 4 bytes like you mention. Why is that?

robincornelius, 2019-09-11:
On Wed, 11 Sep 2019 at 11:21, DaanConn <notifications@github.com> wrote:
>
> Thank you Robin, that makes sense. Just one more thing, the data size should be bigger than 4, so 5 bytes or more, however, in case of an abort, the error code is 4 bytes like you mention. Why is that?

the data size must be >=4, having just 4 is OK ...
 if(SDO_C == NULL || dataRx == 0 || dataRxSize < 4)

There are multiple types of SDO transfer, if the data is 4 bytes or
less it uses what is known as an expedited transfer, that is, in the
SDO request there are 4 empty bytes and the data or abort code can fit
in that, making the SDO handshake a 2 paket transfer. Abort uses the
expedited transfer to send back its error codes. if the data is longer
that 4 bytes you need to use a multi packet SDO transfer to transfer
all the bytes and this can use segmented or block transfer (which are
two slightly different ways of handshaking the data).

Robin

DaanConn, 2019-09-12:
If 4 bytes is OK, I would expect:
`dataRxSize <= 4`
In stead of :
`dataRxSize < 4`

martinwag, 2019-10-08:
`dataRxSize < 4` leads to abort, so
`dataRxSize <= 4` will always fail.

closed 2020-01-27 by CANopenNode