CANopenNode/CANopenNode #114: SDO_process rx Buffer clearing

blankm, 2019-10-31:
In SDO_process, when the statemachine is entered (ln. 903, CO_SDO.c), the first block looks like this:
``` 
/* return immediately if still idle */
        case(CO_SDO_ST_IDLE):{
            return 0;            
        }
```

But before the statemchine ist says: 
`/* state machine (buffer is freed (CLEAR_CANrxNew()) at the end) */`

Well, seems like this is not true for the IDLE case. 

Is this a bug or can't the statemachine be entered in IDLE and this part is never reached in case "IDLE"? Didn't fully run through the code yet.

martinwag, 2019-11-06:
I've had a short look at this, I think it's fine. The comment only applies to the code _afterwards_. The code before has the clear statement on all transitions to idle.

However, something like 

```
    /* return immediately if still idle */
    if(state == CO_SDO_ST_IDLE){
        CLEAR_CANrxNew(SDO->CANrxNew);
        return 0;
    }
```
would be more clean and less error prone in my opinion.

CANopenNode, 2020-01-27:
I'm don't want to CLEAR_CANrxNew at that place. It seems it is OK. Closing now.

closed 2020-01-27 by CANopenNode