CANopenNode/CANopenNode #115: Documentation generation does not seem to work

ArminBmrt, 2019-11-01:
I am having some problems with CANopenNode which will probably be solved by reading the documentation. Problem is the doxygen creates a very basic page with only 'CANopenNode documentation' showing.

I am working on windows, doxygen 1.8.16 (from http://www.doxygen.nl/download.html).

I open doxywizard, open 'Doxyfile' in the canopen node directory and press 'run doxygen'. This gives the almost documentation page. I have attached the doxygen log.

[doxygen_log.txt](https://github.com/CANopenNode/CANopenNode/files/3797243/doxygen_log.txt)

martinwag, 2019-11-04:
The main page is indeed empty. I get a menu tree on the left to select the items, wich works in my case. Have you checked for that?

ArminBmrt, 2019-11-04:
Thanks they indeed in the menu tree. Thanks

closed 2019-11-04 by ArminBmrt