CANopenNode/CANopenNode #117: Code formatting

odesenfans, 2019-11-10:
Hello all,

An issue that came up during the development of my latest patches is the formatting of the code. It seems to me that the formatting currently used, especially spacing, is:

- uncommon/a bit weird
- not enforced properly in a lot of places
- mixes tabs and spaces
- etc

I don't want to start a flame war as this is always a sensitive topic. I just feel like the current formatting is getting in the way of contributing more efficiently. On most projects I work on I use clang-format + a default configuration (Google or LLVM coding style). This is generally "good enough" for everyone, makes it possible to format the code automatically and check that PRs respect the formatting immediately.

Is this of interest?

wilkinsw, 2019-11-10:
I 100% support a formatting pass.  Janez gave us a [codingStyle](https://github.com/CANopenNode/CANopenNode/blob/master/codingStyle) document but it doesn't cover everything.  What seems especially ambiguous is how to handle the spacing of #if guards.  If you go through CANopen.c you'll find pretty much every variation:

- No indent (https://github.com/CANopenNode/CANopenNode/blob/master/CANopen.c#L205)
- Full indent (https://github.com/CANopenNode/CANopenNode/blob/master/CANopen.c#L134)
- Half indent (https://github.com/CANopenNode/CANopenNode/blob/master/CANopen.c#L253)

I think some rules for that sort of thing could be in order

wilkinsw, 2019-11-10:
While going through CANopen.c/h I came up with some examples of scenarios where we could really use some style guideliens.

1. Guarding includes/variable definitions (e.g. CANopen.c#L59)
```c
#ifndef CO_USE_GLOBALS
    #include <stdlib.h> /*  for malloc, free */
    static uint32_t CO_memoryUsed = 0; /* informative */
#endif
```

2.  Nesting multiple guards (e.g. CANopen.c#L74)
```c
#if CO_NO_TRACE > 0
    static uint32_t            *CO_traceTimeBuffers[CO_NO_TRACE];
    static int32_t             *CO_traceValueBuffers[CO_NO_TRACE];
  #ifdef CO_USE_GLOBALS
  #ifndef CO_TRACE_BUFFER_SIZE_FIXED
    #define CO_TRACE_BUFFER_SIZE_FIXED 100
  #endif
  #endif
#endif
```

3. Guarding a prototype (e.g. CANopen.h#L173)
```c
#if CO_NO_NMT_MASTER == 1
    CO_ReturnError_t CO_sendNMTcommand(CO_t *CO, uint8_t command, uint8_t nodeID);
#endif
``` 

4. Guardian a function (e.g. CANopen.c#L415)
```c
/******************************************************************************/
#if CO_NO_LSS_SERVER == 1
CO_ReturnError_t CO_LSSinit(
        uint8_t                 nodeId,
        uint16_t                bitRate)
{
    CO_LSS_address_t lssAddress;
    CO_ReturnError_t err;

    lssAddress.identity.productCode = OD_identity.productCode;
    lssAddress.identity.revisionNumber = OD_identity.revisionNumber;
    lssAddress.identity.serialNumber = OD_identity.serialNumber;
    lssAddress.identity.vendorID = OD_identity.vendorID;
    err = CO_LSSslave_init(
            CO->LSSslave,
            lssAddress,
            bitRate,
            nodeId,
            CO->CANmodule[0],
            CO_RXCAN_LSS,
            CO_CAN_ID_LSS_SRV,
            CO->CANmodule[0],
            CO_TXCAN_LSS,
            CO_CAN_ID_LSS_CLI);

    return err;
}
#endif /* CO_NO_LSS_SERVER == 1 */
```

5. Guarding struct members (e.g. CANopen.h#L131)
```c
/**
 * CANopen stack object combines pointers to all CANopen objects.
 */
typedef struct{
    CO_CANmodule_t     *CANmodule[1];   /**< CAN module objects */
    CO_SDO_t           *SDO[CO_NO_SDO_SERVER]; /**< SDO object */
    CO_EM_t            *em;             /**< Emergency report object */
    CO_EMpr_t          *emPr;           /**< Emergency process object */
    CO_NMT_t           *NMT;            /**< NMT object */
    CO_SYNC_t          *SYNC;           /**< SYNC object */
    CO_RPDO_t          *RPDO[CO_NO_RPDO];/**< RPDO objects */
    CO_TPDO_t          *TPDO[CO_NO_TPDO];/**< TPDO objects */
    CO_HBconsumer_t    *HBcons;         /**<  Heartbeat consumer object*/
#if CO_NO_LSS_SERVER == 1
    CO_LSSslave_t      *LSSslave;       /**< LSS server/slave object */
#endif
#if CO_NO_LSS_CLIENT == 1
    CO_LSSmaster_t     *LSSmaster;      /**< LSS master/client object */
#endif
#if CO_NO_SDO_CLIENT != 0
    CO_SDOclient_t     *SDOclient[CO_NO_SDO_CLIENT]; /**< SDO client object */
#endif
#if CO_NO_TRACE > 0
    CO_trace_t         *trace[CO_NO_TRACE]; /**< Trace object for monitoring variables */
#endif
}CO_t;
```

odesenfans, 2019-11-10:
I propose to keep it simple: no indentation whatsoever for preprocessor conditions, and force a comment after each `#endif` to give a reminder of the condition they're closing.

odesenfans, 2019-11-10:
Here is an example branch I just put up using a typical clang-format configuration:

https://github.com/odesenfans/CANopenNode/blob/use-clang-format/CANopen.h
https://github.com/odesenfans/CANopenNode/blob/use-clang-format/CANopen.c

martinwag, 2019-11-11:
Hi,

Code formatting is always a hot topic. You have the entire range from "most important thing that defines code quality" down to "don't care, just let the auto-formatter do it's work after opening a file"...

This is my opinion about this:
- reformatting breaks usefulness of "git blame" from that point backward. This is one of the most important bug-finding tool for me.
- we will not find one coding style to make everyone happy. I'm used to company modified Linux kernel coding style. The current one is at least weird to everyone :-) .
- The same goes for preprocessor #ifdef, but more extreme. You will always find a place where putting #ifdef with indentation makes something easier to understand.
- Overall, I think it's important that one can break the coding style if this makes something easier to understand. Code formatting is there to help me, not to fulfil some stupid requirement.

regards,
Martin

odesenfans, 2019-11-11:
Hi,

I agree with your point, I'm definitely one to let the auto-formatter do its thing! Answering to your different points:

* Totally agree on the git-blame breakage, it's usually a pain in the ass when projects do a formatting pass.
* Completely don't care, love the approach of python Black or gofmt, formatters with no arguments FTW!
* True as well. A good code formatter can be selectively run on a range of lines. Furthermore there's git-clang-format.
* Agree again.

My take on it is that the current formatting is a pain for the following reasons:
* Variable-length whitespaces make auto-replaces a pain
* Any decent editor or IDE is configured to remove extra whitespace. This pollutes diffs for PRs
* The current formatting makes the project look old/deprecated and IMO gives a bad first impression.
* Since the project is moving again and we'll do some major refactorings (remove drivers, etc), I think it's the ideal time for a formatting pass.

Best regards,
Olivier

schneidersoft, 2019-12-29:
Why not make your life simple and just use ident?
````
# makefile target
format:
        ident -linux *.c stack/*.c .....
````

CANopenNode, 2020-01-10:
I agree with critics, CANopen.c looks terrible, but files in stack directory don't look so bad to me in general. I checked also CO_LSS* files contributed by @martinwag, and they don't differ much from the rest of the code. I'm certainly not for running auto-formatter on the whole project.

However, my codingStyle file may be incomplete, so we can still define specification for auto-formatter. It should be similar to current formatting.

I quickly tested  both, `clang-format` and `indent`. There are many options.
https://github.com/odesenfans/CANopenNode/blob/use-clang-format/CANopen.c by @odesenfans seems OK to me. Which style was used for it?

odesenfans, 2020-01-10:
Hi,

I used clang-format, the LLVM format style + a few custom options I like.

You can see the clang-format file here: https://github.com/odesenfans/CANopenNode/blob/use-clang-format/.clang-format

Olivier

CANopenNode, 2020-01-11:
I tested different options. Your style seems nice, but differs much from "standard" LLVM.

I get very similar result with simple `indent` command, Linux style and indentation of 4 spaces:
```
indent --linux-style --no-tabs --indent-level4
```
This style is quite identical to coding style I tried to follow, except it has some more spaces in some lines.

`indent` command is also easy to use in simple text editor, such as Geany, by just selecting segment of code and hit `Ctrl+1` (set options in Edit->Format->Send Selection).

odesenfans, 2020-01-11:
Up to you, the advantage with a .clang-format file is that clang-format looks for the file automatically from any folder in the project. It’s also well integrated with IDEs and editors (CLion and Atom support it very well). I prefer clang-format from habit, but the most important thing is to have a formatter. Doesn’t really matter which one!

Olivier

CANopenNode, 2020-01-11:
Actually it doesn't matter which formatter to use. We can specify mandatory requirements for each.
For example:
- clang-format:
```
BasedOnStyle: LLVM
IndentWidth: 4
BinPackArguments: false
BinPackParameters: false
```
- indent:
```
indent --linux-style --no-tabs --indent-level4
```
We can add some extra specifications, so they will match in general. We can also include `.clang-format` file, but it should produce similar results as `indent`

odesenfans, 2020-01-11:
Sounds good to me!

CANopenNode, 2020-01-14:
Code formatting is now defined more precisely inside file `.clang-format`.

closed 2020-01-14 by CANopenNode