CANopenNode/CANopenNode #118: inttypes.h not available for Microchip XC16 compiler

trojanobelix, 2019-11-11:
CO_trace.c now includes <inttypes.h> to use %PRIu32 data type instead of %lu for printf function.
Unfortunately the Microchip XC16 compiler for dsPIC30/dsPIC33 has no inttypes.h implementation.
That leads to a fatal compile error:
 
**../CANopenNode/stack/CO_trace.c:48:22: fatal error: inttypes.h: No such file or directory**

Example of the code change:
`/* Different functions for printing points for different data types. */
static uint32_t printPointCsv(char *s, uint32_t size, uint32_t timeStamp, int32_t value) {
+    return snprintf(s, size, "" **%PRIu32** ";%" PRId32 "\n", timeStamp,              value);
-    return snprintf(s, size, "%lu;%ld\n", timeStamp,             value);
}`

JuPrgn, 2019-11-11:
For me there was already an error about snprintf with this same compiler. This is not the case for you ?

trojanobelix, 2019-11-11:
I'm not sure if I understand your question.
The dsPICS as the other 16 bit processors of Microchip uses the XC16 compiler. They compiled fine until the data types in snprintf changes to PRIu32, which are normally defined in inttypes.h. But this header does not come with the XC16, only with the XC32 (Microchips 32 bit compiler for PIC32)

Both compiler based on gcc.

JuPrgn, 2019-11-11:
Ok I had already compilation error with snprintf and xc16 before this change.

wilkinsw, 2019-11-11:
I think the simplest solution would be to guard CO_trace.c/h with `#if CO_NO_TRACE > 0` so the code isn't compiled if trace isn't being used (the default).

wilkinsw, 2019-12-20:
@trojanobelix did #137 fix your issue?

JuPrgn, 2019-12-20:
Does this fix this issue ? 
We could also revert the commit to keep CO_Trace compatible with XC16 and similar compilers that implements snprintf but doesn't have inttypes.h ?

JuPrgn, 2019-12-20:
@wilkinsw did not seen you posted too 😄

trojanobelix, 2020-01-02:
I have no problem, with the changes in #137, but it does not fix this problem.

The microchip XC16 compiler for the dsPIC33 driver doesn't define 
uintptr_t, intptr_t, PRIu32 and other "modern" stuff that came with C99 and later.

Therfore using **PRIu32** instead of **lu** in printf generates an error.

It is more  a Microchip problem, as they do not add support for a 20 year old standard in a currently sold compiler.

But it is unlikley that microchip will change that. It might be the best way to declare the missing types to the dsPIC driver files should work as those are included in everything.

wilkinsw, 2020-01-02:
What if you download inttypes.h and include it with your source?  If it doesn't exist already there would be no conflicts.

trojanobelix, 2020-01-06:
Yes, I took inttypes.h from
https://github.com/openbsd/src/blob/master/include/inttypes.h
and it works.

wilkinsw, 2020-01-08:
@trojanobelix is your issue resolved, then?

trojanobelix, 2020-01-10:
Yes, the issue is solved. 

But it might be helpful for newbies if the example projects in
CANopenPIC (CANopen demo implementation for PIC microcontrollers)
could be compiled with latest XC compiler and latest CANopenNode

closed 2020-01-10 by trojanobelix

CANopenNode, 2020-01-22:
I added three lines into CO_driver_target.h for dsPIC3x:
```
#define CO_OWN_INTTYPES
#define PRIu32 "lu"
#define PRId32 "ld"
```