CANopenNode/CANopenNode #123: SDO download for large blob fails

geoffrey-vl, 2019-11-14:
I've setup OD object 1F50 for program download.
The target node is a STM32L4, the master node a linux with quadcore GHz processor.

I've successfully send 10Kb payloads until now, so the configuration is OK.
When sending a bigger sized blob, roughly 100Kb, the SDO download fails.
It goes well until roughly 30%, but than it "hangs".

What happens is that at the slave node we don't receive anything in the OD 1F50 its callback handler. On the master side I'm using canopencomm/canopend. Canopencomm takes the large SDO write command from a file. During sending the command doesn't return, which is to be expected since it should only return after the entire 100Kb SDO write has been completed.

Using candump on the master's side I've noticed that at the sudden point where things go wrong, the master decides to send all "0"s. Here is a screenshot of candump:

[image]

Part of the data that needs to be send at this stage doesn't particularly look fishy.
The data just before it starts failing (, as also seen in the screenshot):
`F3 05 83 33 B1 00 20 6F F0 05 01 00 F0 81 F8 00 20 08 BD`

Here is that part where it starts failing:
 `FF F7 BA FE 00 28 F4 D1 10 46 DF`

I assumed maybe the `FF` could be troublesome, however it did successfully transmit other `FF` many times before. When I send the same data again, it fails at exactly the same point, so I can easily repeat the issue. When I remove some data more to the front of the command file, the fail happens few % earlier. If I remove some data from the back of the command file, the fail also happens few % earlier. So it's not tied to the actual data I'd guess, but really more like at a sudden point. What I've also tried is increasing by tenfold the interval time in `CO_master` of `canopend`:

```
    /* Download data. Loop in 5 ms intervals. */
    if(err == 0){
        CO_SDOclient_return_t ret;
        uint16_t timer1msPrev;
        struct timespec sleepTime;

        timer1msPrev = CO_timer1ms;
        sleepTime.tv_sec = 0;
        sleepTime.tv_nsec = 50000000;
```

It changes nothing, neither does improving the latency in OD1F50 callback at the slave side.

However, when I change `canopencomm` to use block transfer (`static uint8_t              blockTransferEnable = 1;`
=> it succeeds!

odesenfans, 2019-11-14:
A bit hard to see what could go wrong without a bit more information / the implementation of your callback handler for `0x1F50`.

Just an uneducated guess, but this sounds more like a host issue than a MCU issue. I'd advise to debug canopencomm/canopend to figure out what's the condition that suddenly causes it to send zeroes.

geoffrey-vl, 2019-11-14:
The driver (and base implementation of OD 1F50) is out of tree and can be found [here](https://github.com/Alphatronics/mbed-os-example-canopen/blob/fwupgrade/OD/CO_fwupgrade.cpp).

odesenfans, 2019-11-14:
I see you have some debug logs in the 1F50 handler. Could you attach the output coming from the MCU?

geoffrey-vl, 2019-11-14:
Well with the debugger attached, once the 0's start to appear I've added a breakpoint in the 1F50 handler and it doesn't get hit, so the handler is no longer called at this stage. I'm also sure that each time the handler was being called before the problem occurs, the function executes just fine (i.e. doesn't lock the device somehow).
I guess there is not much we can learn from logs I currently have. My best bet for now is that somehow someone is doing the same thing, that would at least help to rule out some options (or maybe not)

martinwag, 2019-12-09:
Is this fixed using https://github.com/CANopenNode/CANopenSocket/pull/14 ?

geoffrey-vl, 2019-12-09:
No, that fixes some other HEAP problem I had in CANOpenSocket while using it on my embedded system.

Manfredi9, 2020-08-27:
Hi, @geoffrey-vl I have the same issue, did you fix it?

JuPrgn, 2020-08-27:
@Manfredi9 just a hint, you could try to change CO_SDO_BUFFER_SIZE to the maximum value 889. I wonder if this can have an effect on this issue.

Manfredi9, 2020-08-28:
@JuPrgn Thanks for the hint, but it didn't fix the problem...

JuPrgn, 2020-08-28:
Ok I successfully send 260kb using my bootloader with max buffer size and never had any issue. Could you share some details on what are you doing ?

Manfredi9, 2020-08-28:
@JuPrgn another problem is that I can't send the firmware with SDO domain when I use "./canopencomm set sdo_block 1".
I receive : [1] ERROR: 0x08000000 - General error.
                 canopend generic error: 0x15800068

It starts sending only if "./canopencomm set sdo_block 0".

JuPrgn, 2020-08-28:
I cannot help you with canopencomm as I don't use it but I could help if the problem is on your CANopenNode configuration. Maybe there is a bug on canopencomm ? Did you try other tools ?

geoffrey-vl, 2020-08-29:
> Hi, @geoffrey-vl I have the same issue, did you fix it?

No, I'm relying in block transfers where the issue didn't occur for me. I'm also on 1.3 branch.

CANopenNode, 2020-12-28:
I'm not quite sure, how is with this issue now. Currently we have:
- new Object Dictionary approach, rewritten and tested SDO server in CANopenNode version 4.0
- rewritten and tested SDO client in CANopenNode version 2.0 and above
- no additional information for old Object Dictionary and old SDO server.
I'm closing this issue for now. If there will be similar problems, new issue should be opened.

closed 2020-12-28 by CANopenNode